mavlink/MAVSDK #2188: set_rate_attitude(): ignores requested rate

ygratz, 2023-11-26:
Hi,
Working with SITL, I see that as the SITL starts working it sends various messages at given rates.
For example the attitude message is sent every 250ms.

When I run:
telemetry.set_rate_attitude(1.0);

It does not have any affect on the message rate.
So is true for other "set rates".

What do I need to do to overcome this?
Thanks

julianoes, 2023-11-26:
Is this PX4?

ygratz, 2023-11-26:
arducopter

Jai-GAY, 2023-11-27:
> telemetry.set_rate_attitude(1.0);

1/4=0.25=250ms, does 1 = 250ms?

julianoes, 2023-11-27:
Ok, I need to try to reproduce this with ArduCopter? Can I reproduce it with just SITL?

Jai-GAY, 2023-11-27:
> Can I reproduce it with just SITL?

@ygratz , do you have any comment? for my cases, I am using Arducopter on cubeorange. I am not sure can it be reproduced using SITL.

julianoes, 2024-05-20:
Is this still an issue?

ygratz, 2024-05-20:
No – it is not an issue anymore.
This behavior was because I connected using MAXProxy, which sets the messages rates by itself.

"Organization", "name": "GitHub", "url": "https://github.com" } } ]

Default Profile
*********************************************************************************************** Please consider the environment before printing this email ! The information contained in this communication is proprietary to Israel Aerospace Industries Ltd. and/or third parties, may contain confidential or privileged information, and is intended only for the use of the intended addressee thereof. If you are not the intended addressee, please be aware that any use, disclosure, distribution and/or copying of this communication is strictly prohibited. If you receive this communication in error, please notify the sender immediately and delete it from your computer. Thank you. Visit us at: www.iai.co.il

julianoes, 2024-05-20:
Aha, thanks for the note!

closed 2024-05-20 by julianoes