mavlink/MAVSDK #2190: unix_epoch_time subscription not working in C++ and Python

LarsLeferenz, 2023-12-06:
Hello, 

I'm currently trying to sync time between an external application and my mavlink device in C++. To do this I'm trying to subscribe to the "unix_epoch_time" telemetry.
Sadly, the callback is never called. To explore the issue, I also created a simple python script to display the telemetry message:
```python
import asyncio
from mavsdk import System

async def run():
    # Connect to the MAVSDK system
    system = System(mavsdk_server_address='localhost', port=50051)
    await system.connect()

    await system.telemetry.set_rate_unix_epoch_time(1)
    # Subscribe to time updates
    async for time in system.telemetry.unix_epoch_time():
        print(f"Time: {time}")

if __name__ == "__main__":
    asyncio.run(run())
```
Python isn't outputting anything. The Mavsdk server output is as follows:
```
PS C:\Users\Lars\Downloads> .\mavsdk_server_win32.exe -p 50051 serial://COM5:57600
[04:46:05|Info ] MAVSDK version: v1.4.17 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:20)
[04:46:05|Info ] Waiting to discover system on serial://COM5:57600... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
[04:46:05|Debug] New: System ID: 1 Comp ID: 1 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:496)
[04:46:05|Debug] Component Autopilot (1) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:377)
[04:46:05|Warn ] Vehicle type changed (new type: 2, old type: 0) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:225)
[04:46:05|Debug] Discovered 1 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:578)
[04:46:05|Info ] System discovered (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:63)
[04:46:05|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:53)
[04:46:05|Info ] Server set to listen on 0.0.0.0:50051 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:54)
```

I also looked into the mavlink messages through QGroundControl and I am receiving SYSTEM_TIME and TIME_SYNC messages:
[image]

My FC is a Cube Orange running PX4 v1.14 connected via USB. I'm using mavsdk-python 1.4.9., mavsdk-server 1.4.17 and I my C++ solution utilized v1.4.16.

I should also mention that I don't have a GPS connected at the moment.

Am I using unix_epoch_time wrong or do I have to set a parameter in PX4 perhaps?

julianoes, 2023-12-10:
It looks like the `UnixEpochTime` method is based on [UTM_GLOBAL_POSITION](https://mavlink.io/en/messages/common.html#UTM_GLOBAL_POSITION).time.

https://github.com/mavlink/MAVSDK/blob/v1.4/src/mavsdk/plugins/telemetry/telemetry_impl.cpp#L1328-L1343

So this probably means it only works with GPS working.

Are you trying to get UTC time, or boot time? I assume UTC?

LarsLeferenz, 2023-12-10:
Ah ok,
I'm trying to get the systems UTC Time.

julianoes, 2024-05-20:
Fixed with #2294.

closed 2024-05-20 by julianoes