mavlink/MAVSDK #2191: When multiple vehicles are connected, all systems get data from all vehicles, regardless of sysid.

poli0048, 2023-12-08:
I am testing a program that will control multiple vehicles at once. For testing I have two drones connected: one real drone over a telemetry radio and a second, simulated drone using a Docker container. The drones have been assigned different system IDs, and when I look at mavsdk.systems() I see two systems. The documentation suggests that when you create a plugin (like Telemetry) for a system it will give you data just for that system. Similarly, if you register callbacks with a plugin that was created for a system, it should fire and provide data only for that system. This does not appear to be what happens. If I create one Telemetry object for each system, they both get all data from both drones. This means that both Telemetry objects show, for instance, battery levels that toggle back and forth between the values of the two drones.

I have traced through the code and it looks like the issue is related to some changes in mavsdk_impl.cpp in function:
void MavsdkImpl::receive_message(mavlink_message_t& message, Connection* connection) {}

Towards the end of this function there are some remnants of routing logic. It currently looks like this:
for (auto& system : _systems) {
        if (system.first == message.sysid) {
            // system.second->system_impl()->process_mavlink_message(message);
            mavlink_message_handler.process_message(message);
            break;
        }
}

The commented out line looks like the message would be routed only to a system that matches the sysid of the incoming message. The line that replaces this just shovels the message to a generic handler, which appears to call all registered callbacks regardless of the sysid of the system that registered them. This would seem to explain the behavior I am seeing.  It looks like commit ad2f80b by [julianoes](https://github.com/julianoes) on May 9, 2022 is responsible for this change but I don't see any explanation behind this decision.

Is this the intended behavior? If so it seems interacting with multiple systems through one MAVSDK-based program is not possible anymore. Any thoughts or insights would be appreciated.

JonasVautherin, 2023-12-09:
> The documentation suggests that when you create a plugin (like Telemetry) for a system it will give you data just for that system. 

Yes it should do that.

Which version of MAVSDK are you using?

poli0048, 2023-12-09:
I am using the master branch as of October 20, 2023. On commit a9c198156722b26cbbcadfb148d01116deea4f2f.

JonasVautherin, 2023-12-09:
And how did you establish that this commit from 2022 is responsible for this change? Did you manage to revert/fix it?

Maybe @julianoes has some insights about these changes, too.

poli0048, 2023-12-09:
Hi Jonas,

This looks like the commit that commented out the line:
system.second->system_impl()->process_mavlink_message(message);
and replaced it with the line:
mavlink_message_handler.process_message(message);

The line being replaced in this commit seems to route messages only to systems that match the sysid of the message. The new line does not look like it does this check, and when I look at the code in the MavlinkMessageHandler class it doesn't look like it distinguishes between callback registrations based on originating system and the sysid doesn't get used for deciding which callbacks to invoke for a given message. In other words, before this commit it looks like the intent was to route messages only to systems with matching sysid. After this commit it looks like all messages are sent to all systems, assuming they match the sysid of at least one existing system. I have not tried reverting to before this commit though since it is from about 1.5 years ago.

JonasVautherin, 2023-12-09:
So you're talking about [this](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/mavsdk_impl.cpp#L394-L400), right?

```cpp
    for (auto& system : _systems) {
        if (system.first == message.sysid) {
            // system.second->system_impl()->process_mavlink_message(message);
            mavlink_message_handler.process_message(message);
            break;
        }
    }
```

So before, the `process_message` was done at the system level, and the `if (system.first == message.sysid) {` would ensure that the system would process the message only if `message.sysid` was corresponding. Now it is not the case anymore, it will always call `mavlink_message_handler.process_message(message);` for the messages.

I only quickly looked there, but maybe [this function](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/mavlink_message_handler.cpp#L55) should somehow check the system id? Maybe check that the entry corresponds to the system from which the message is coming from?

julianoes, 2023-12-10:
Oh dear, did I break this? Thanks for reporting @poli0048!

poli0048, 2023-12-11:
Hi Jonas,

I think we are reading the code the same way - I agree with your assessment. It seems to me that there are two options:
1 - Revert to the previous architecture where void MavsdkImpl::receive_message() does sysid-based message routing, but then each system processes the messages it receives separately (and presumably invokes necessary callbacks).
2 - Modify the MavlinkMessageHandler class so that when callbacks are registered, the sysid of the registering system is recorded in the callback table, then only pass messages to systems if the callback was registered with the same sysid as the incoming message.

I am still getting familiar with MAVSDK and I don't know the reasons for the changes that were previously made here, so I can't say which approach is necessarily better, but I think either strategy could resolve this issue.

JonasVautherin, 2023-12-11:
I would tend to go for 2. @julianoes do you have an opinion on that?

@poli0048 would you be willing to try fixing it (once we agree with Julian on what would be the way to go)?

poli0048, 2023-12-11:
If you would both prefer that I take a crack at fixing it I can try. I will wait for Julian to weigh in.

julianoes, 2023-12-15:
I'm looking into it now. You're spot on, and I don't quite remember my thinking back then either... and I'm also tending to 2.

julianoes, 2023-12-16:
@poli0048 can you try https://github.com/mavlink/MAVSDK/pull/2192 and check if it fixes the problem? That would be great!

julianoes, 2023-12-16:
Sorry for just doing it but it bugged me enough that I wanted to do it rightaway.

julianoes, 2023-12-16:
Looks like I broke the tests...

julianoes, 2023-12-16:
I first tried 2. but it turns out that didn't work for commands. The reason was that the initial system gets initialized with target 0. At that point the MavlinkCommandSender registers for any messages, in this case for 0. Then, once a system is discovered, the system ID is set to 1 but the MavlinkCommandSender's subscription is still 0, and it doesn't receive the acks.

I therefore went back to 1, so we now have MavlinkMessageHandlers in MavsdkImpl for components and in SystemImpl for systems.

closed 2023-12-16 by julianoes

poli0048, 2023-12-19:
I know this issue is closed now but I wanted to follow up - I have just tested the changes you made and it does seem to resolve the issue for me. I can connect to two drones now and the messages are properly separated by system. Thank you for taking the time to fix this!

julianoes, 2023-12-20:
Awesome, thanks for the feedback @poli0048 and raising this blunder in the first place!

PR #2192 (2023-12-16): core: fix incoming messages by sysid
This fixes a recent regression after a refactoring where messages from different systems are no longer filtered according to the source system ID and instead are distributed to all systems.

Fixes #2191.