mavlink/MAVSDK #2207: heap-use-after-free after heartbeats timed out when using UDP connection

keith-bennett-airmap, 2024-01-10:
# tl;dr

This appears to be a timing issue and the application typically crashes at intermittent intervals. Shared or overused compute clusters can cause delays to network traffic (especially with a cloud-based simulation). Without directly observing a packet capture, this is a proposed sequence of events to reproduce in a regression test:
- connect over udp
- stop sending heartbeats
- get notified of disconnect
- immediately send gimbal events
- crash

# Description
This is observed with [mavsdk 7fe1b6083545a0760dc306388ea946ded78316da](https://github.com/mavlink/MAVSDK/tree/7fe1b6083545a0760dc306388ea946ded78316da) when compiled with Address Sanitizer.

mavsdk library is used by a DroneUp application, the `mavlink_shim` and the whole application is built in a Docker-based workflow. We see a heap-use-after-free error which we believe is related to mavsdk and UDP network connectivity. The issue's reproducibility is intermittent -- we don't see it frequently nor for every heartbeats timed out event. Debugging an intermittent issue is difficult :rofl: 

We observe these events:
- our `mavlink_client` object is using mavsdk. It's configured to use a udp connection and has successfully connected.
- At some point, mavsdk reports a connection loss (eg, missed heartbeats)
- We then observe a crash in the gimbal manager inside of mavsdk
- The crash appears to be within mavsdk's internal components.

The crash provides the following stack trace information.

# Address Sanitizer Defect Report
```
=================================================================
==1==ERROR: AddressSanitizer: heap-use-after-free on address 0x6080000103e0 at pc 0x563c693f2add bp 0x7f6c3d6f0d50 sp 0x7f6c3d6f0d40
READ of size 8 at 0x6080000103e0 thread T17
    #0 0x563c693f2adc in std::_Function_base::_M_empty() const /usr/include/c++/11/bits/std_function.h:247
    #1 0x563c69467a55 in std::function<void (mavsdk::Gimbal::ControlStatus)>::operator bool() const /usr/include/c++/11/bits/std_function.h:573
    #2 0x563c698e4a32 in mavsdk::GimbalProtocolV2::process_gimbal_manager_status(__mavlink_message const&) (/usr/src/app/install/bin/mavlink_shim+0x6cea32)
    #3 0x563c698e61bf in mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}::operator()(__mavlink_message const&) const (/usr/src/app/install/bin/mavlink_shim+0x6d01bf)
    #4 0x563c698e981c in void std::__invoke_impl<void, mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}&, __mavlink_message const&>(std::__invoke_other, mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}&, __mavlink_message const&) (/usr/src/app/install/bin/mavlink_shim+0x6d381c)
    #5 0x563c698e8c94 in std::enable_if<is_invocable_r_v<void, mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}&, __mavlink_message const&>, void>::type std::__invoke_r<void, mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}&, __mavlink_message const&>(mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}&, __mavlink_message const&) (/usr/src/app/install/bin/mavlink_shim+0x6d2c94)
    #6 0x563c698e7821 in std::_Function_handler<void (__mavlink_message const&), mavsdk::GimbalProtocolV2::control_async(std::function<void (mavsdk::Gimbal::ControlStatus)>)::{lambda(__mavlink_message const&)#1}>::_M_invoke(std::_Any_data const&, __mavlink_message const&) (/usr/src/app/install/bin/mavlink_shim+0x6d1821)
    #7 0x563c69836a42 in std::function<void (__mavlink_message const&)>::operator()(__mavlink_message const&) const (/usr/src/app/install/bin/mavlink_shim+0x620a42)
    #8 0x563c69836539 in mavsdk::MavlinkMessageHandler::process_message(__mavlink_message const&) (/usr/src/app/install/bin/mavlink_shim+0x620539)
    #9 0x563c697b222f in mavsdk::MavsdkImpl::receive_message(__mavlink_message&, mavsdk::Connection*) (/usr/src/app/install/bin/mavlink_shim+0x59c22f)
    #10 0x563c697b2ba1 in mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}::operator()(__mavlink_message&, mavsdk::Connection*) const (/usr/src/app/install/bin/mavlink_shim+0x59cba1)
    #11 0x563c697b972e in void std::__invoke_impl<void, mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}&, __mavlink_message&, mavsdk::Connection*>(std::__invoke_other, mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}&, __mavlink_message&, mavsdk::Connection*&&) (/usr/src/app/install/bin/mavlink_shim+0x5a372e)
    #12 0x563c697b9140 in std::enable_if<is_invocable_r_v<void, mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}&, __mavlink_message&, mavsdk::Connection*>, void>::type std::__invoke_r<void, mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}&, __mavlink_message&, mavsdk::Connection*>(mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}&, __mavlink_message&, mavsdk::Connection*&&) (/usr/src/app/install/bin/mavlink_shim+0x5a3140)
    #13 0x563c697b8ca4 in std::_Function_handler<void (__mavlink_message&, mavsdk::Connection*), mavsdk::MavsdkImpl::add_udp_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, mavsdk::ForwardingOption)::{lambda(__mavlink_message&, mavsdk::Connection*)#1}>::_M_invoke(std::_Any_data const&, __mavlink_message&, mavsdk::Connection*&&) (/usr/src/app/install/bin/mavlink_shim+0x5a2ca4)
    #14 0x563c699b2792 in std::function<void (__mavlink_message&, mavsdk::Connection*)>::operator()(__mavlink_message&, mavsdk::Connection*) const (/usr/src/app/install/bin/mavlink_shim+0x79c792)
    #15 0x563c699b232e in mavsdk::Connection::receive_message(__mavlink_message&, mavsdk::Connection*) (/usr/src/app/install/bin/mavlink_shim+0x79c32e)
    #16 0x563c698549ec in mavsdk::UdpConnection::receive() (/usr/src/app/install/bin/mavlink_shim+0x63e9ec)
    #17 0x563c69856225 in void std::__invoke_impl<void, void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*>(std::__invoke_memfun_deref, void (mavsdk::UdpConnection::*&&)(), mavsdk::UdpConnection*&&) (/usr/src/app/install/bin/mavlink_shim+0x640225)
    #18 0x563c69856178 in std::__invoke_result<void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*>::type std::__invoke<void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*>(void (mavsdk::UdpConnection::*&&)(), mavsdk::UdpConnection*&&) (/usr/src/app/install/bin/mavlink_shim+0x640178)
    #19 0x563c698560d8 in void std::thread::_Invoker<std::tuple<void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) (/usr/src/app/install/bin/mavlink_shim+0x6400d8)
    #20 0x563c6985608d in std::thread::_Invoker<std::tuple<void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*> >::operator()() (/usr/src/app/install/bin/mavlink_shim+0x64008d)
    #21 0x563c6985606d in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (mavsdk::UdpConnection::*)(), mavsdk::UdpConnection*> > >::_M_run() (/usr/src/app/install/bin/mavlink_shim+0x64006d)
    #22 0x7f6c4770b252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252)
    #23 0x7f6c47394ac2 in start_thread nptl/pthread_create.c:442
    #24 0x7f6c4742665f  (/lib/x86_64-linux-gnu/libc.so.6+0x12665f)
[... cut ...]
    2. It was captured with an unsynchronized `this` [here](https://github.com/mavlink/MAVSDK/blob/7fe1b6083545a0760dc306388ea946ded78316da/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.cpp#L289-L291) for the callback.
    3. The function pointer is assigned a value [here](https://github.com/mavlink/MAVSDK/blob/7fe1b6083545a0760dc306388ea946ded78316da/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.cpp#L21-L53) during asynchronous gimbal control.

UDP, being connectionless, still processes received datagrams even though the _event thread_ is handling a disconnect event and deletes things that the _socket thread_ needs for continued processing.

julianoes, 2024-01-11:
Thanks for the detailed report! Nice debugging work!

My first thought is that we need to add the call to unregister the message subscription in the destructor of `GimbalProtocolV2`.

However, you mention that access of that object is not synchronized, so I wonder if we also need a mutex somewhere. The message registration and calling is already protected/synchronized, so maybe that's enough... 🤔

keith-bennett-airmap, 2024-01-11:
A mutex, perhaps. But the problem is capturing raw `this` which gets deleted. A mutex won't protect from that.

Instead, if the `GimbalProtocolBase` is owned by `shared_ptr`, then the lambda can capture a `w = weak_from_this()` and try to promote using `auto self = w.lock()`. If it succeeded then proceed with `self->process_gimbal_manager_status(message)`, otherwise return immediately.

Using `weak_from_this()` requires `std::enable_shared_from_this`. Best use of `std::enable_shared_from_this` is with non-public constructors. It also means that the object cannot be constructed on the stack, otherwise `weak_from_this()` will return `nullptr`.

I understand the sequence of events but I'm not sure that a proposed solution fixes the problem until a regression test can reliably crash at the same location. I'm not 100% sure how to build a regression test for it.

```diff
diff --git a/src/mavsdk/plugins/gimbal/gimbal_impl.cpp b/src/mavsdk/plugins/gimbal/gimbal_impl.cpp
index ee0e3d22..002e677c 100644
--- a/src/mavsdk/plugins/gimbal/gimbal_impl.cpp
+++ b/src/mavsdk/plugins/gimbal/gimbal_impl.cpp
@@ -6,6 +6,7 @@
 #include <cmath>
 #include <functional>
 #include <thread>
+#include <mutex>
 
 namespace mavsdk {
 
@@ -50,7 +51,7 @@ void GimbalImpl::enable()
 
 void GimbalImpl::disable()
 {
-    _gimbal_protocol.reset(nullptr);
+    _gimbal_protocol.reset();
 }
 
 void GimbalImpl::receive_protocol_timeout()
@@ -58,7 +59,7 @@ void GimbalImpl::receive_protocol_timeout()
     // We did not receive a GIMBAL_MANAGER_INFORMATION in time, so we have to
     // assume Version2 is not available.
     LogDebug() << "Falling back to Gimbal Version 1";
-    _gimbal_protocol.reset(new GimbalProtocolV1(*_system_impl));
+    _gimbal_protocol = GimbalProtocolV1::make(*_system_impl);
     _protocol_cookie = nullptr;
 }
 
@@ -74,8 +75,8 @@ void GimbalImpl::process_gimbal_manager_information(const mavlink_message_t& mes
 
         _system_impl->unregister_timeout_handler(_protocol_cookie);
         _protocol_cookie = nullptr;
-        _gimbal_protocol.reset(new GimbalProtocolV2(
-            *_system_impl, gimbal_manager_information, message.sysid, message.compid));
+        _gimbal_protocol = GimbalProtocolV2::make(
+            *_system_impl, gimbal_manager_information, message.sysid, message.compid);
     }
 }
 
diff --git a/src/mavsdk/plugins/gimbal/gimbal_impl.h b/src/mavsdk/plugins/gimbal/gimbal_impl.h
index 7a7ab1f8..1e1374c9 100644
--- a/src/mavsdk/plugins/gimbal/gimbal_impl.h
+++ b/src/mavsdk/plugins/gimbal/gimbal_impl.h
@@ -59,7 +59,7 @@ public:
     const GimbalImpl& operator=(const GimbalImpl&) = delete;
 
 private:
-    std::unique_ptr<GimbalProtocolBase> _gimbal_protocol{nullptr};
+    std::shared_ptr<GimbalProtocolBase> _gimbal_protocol{nullptr};
 
     void* _protocol_cookie{nullptr};
 
diff --git a/src/mavsdk/plugins/gimbal/gimbal_protocol_base.h b/src/mavsdk/plugins/gimbal/gimbal_protocol_base.h
index 5bb5a4d5..ca64588e 100644
--- a/src/mavsdk/plugins/gimbal/gimbal_protocol_base.h
+++ b/src/mavsdk/plugins/gimbal/gimbal_protocol_base.h
@@ -1,14 +1,21 @@
 #pragma once
 
[... 126 lines cut ...]
                 this);
         }
diff --git a/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.h b/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.h
index 3c2b1d4b..e3b519bf 100644
--- a/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.h
+++ b/src/mavsdk/plugins/gimbal/gimbal_protocol_v2.h
@@ -7,7 +7,17 @@ namespace mavsdk {
 
 class GimbalProtocolV2 : public GimbalProtocolBase {
 public:
+    using shared_ptr = std::shared_ptr<GimbalProtocolV2>;
+    using weak_ptr = std::weak_ptr<GimbalProtocolV2>;
+
+    static shared_ptr make(
+        SystemImpl& system_impl,
+        const mavlink_gimbal_manager_information_t& information,
+        uint8_t gimbal_manager_sysid,
+        uint8_t gimbal_manager_compid);
+
     GimbalProtocolV2(
+        Protected,
         SystemImpl& system_impl,
         const mavlink_gimbal_manager_information_t& information,
         uint8_t gimbal_manager_sysid,
```

julianoes, 2024-01-11:
Hmm, I see, there is more to this...

julianoes, 2024-01-19:
@keith-bennett-airmap can you still reproduce your issue after #2208? I still wonder if that isn't enough.

julianoes, 2024-01-23:
Right, that's good to hear but you're right. We could try to create a system test that excites it that we can then run many times over night with address sanitizer enabled.

julianoes, 2024-05-20:
I have recently fixed some heap-use-after-free in the gimbal plugin. I believe including this one.

https://github.com/mavlink/MAVSDK/pull/2298
https://github.com/mavlink/MAVSDK/pull/2286

closed 2024-05-20 by julianoes