mavlink/MAVSDK #2209: how to get Ardupilot Camera Trigger status?

SeongminJaden, 2024-01-12:
Hello, I am developing software to control drones using MAVSDK c++, but I am having a problem, so I would like to ask a question.

What I want to do is set a flight mission to control the camera trigger at each waypoint of misson, but in mavsdk, I want to do a cout every time the trigger is controlled during the mission. But it's not working well. Please help.

` mavsdk.subscribe_on_message_async(
        MAVLinkMessage::Message::CAMERA_IMAGE_CAPTURED,
        [&](const MAVLinkMessage& message) {
            // 이미지가 캡처되면 실행될 코드
            std::cout << "이미지 캡처 완료\n";

                std::ostringstream filenameStream;
                filenameStream << baseFilename << i << fileExtension;
                std::string filename = filenameStream.str();
                std::string command = "gphoto2 --port=usb: --folder=" + downloadPath +
                    " --capture-image-and-download --filename=\"" + downloadPath + filename + "\"";
                int result_trigger = std::system(command.c_str());
                if (result_trigger == 0) {
                    std::cout << "Image captured and downloaded: " << filename << std::endl;
                }
                else {
                    std::cerr << "Error capturing or downloading image." << std::endl;
                    return 1;
                }
                sleep_for(seconds(5));

                std::string exifCommand = exifCommandTemplate + filePathPrefix + std::to_string(i) + filePathSuffix;
                std::string move_enc = move + filePathPrefix + std::to_string(i) + filePathSuffix + enc;
                std::cout << exifCommand << std::endl;
                // Convert latitude and longitude to strings with full precision
                std::ostringstream latitudeStr, longitudeStr;
                latitudeStr << latitude;
                longitudeStr << longitude;

                // Convert altitude to a string and append "m"
                std::ostringstream altitudeStr;
                altitudeStr << altitude;

                size_t pos = exifCommand.find("xxx");
                exifCommand.replace(pos, 3, latitudeStr.str());
                pos = exifCommand.find("xxx");
                exifCommand.replace(pos, 3, longitudeStr.str());
                pos = exifCommand.find("xxx");
                exifCommand.replace(pos, 3, altitudeStr.str());
                int result = std::system(exifCommand.c_str());
        },
        nullptr);

    trigger_result = action.set_trigger_control(Action::TriggerControl::Disable);
    if (trigger_result != Action::Result::Success) {
        std::cerr << "트리거 제어 비활성화 실패: " << trigger_result << '\n';
        return 1;
    }`

JonasVautherin, 2024-01-12:
First, instead of `subscribe_on_message_async`, you should probably use the Camera plugin ([`subscribe_capture_info`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/camera/camera.proto#L75), which will give you an [`CaptureInfo`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/camera/camera.proto#L323) object for each photo taken.

Then, when you say "it is not working well", what you do mean? What can happen is that the corresponding mavlink message is lost (because it is an unreliable mavlink broadcast), and therefore MAVSDK does not guarantee that the callback is triggered for every single photo taken.

julianoes, 2024-05-27:
No more answers, closing.

closed 2024-05-27 by julianoes