mavlink/MAVSDK #2211: Mission Upload issues with Mavsdk Java and Ardupilot

acofis, 2024-01-12:
In an effort to build capabilities while testing and learning mavsdk I have been modifying some of the example code in MavSDK Java to talk with a SITL Arducopter drone.

My Environment:
	- Windows 11 host
	- Ardupilot SITL using WSL Linux ubuntu
	- Mavsdk server from [Releases](https://github.com/mavlink/MAVSDK/releases/download/v2.0.0/mavsdk_server_musl_x86_64) hosted in Alpine 3.19 docker
	-  MavSDKJava examples running from intellij gradle task

The Issue:
When attempting to start a mission upload using the code snippet below I get a "TIMEOUT" return after just a second or 2. This error happens regardless of using MissionRaw or Mission plugins.

To Reproduce:
	- run mavsdk server  v2.0 using `./mavsdk_server --sysid 10 --compid 10 udp://:14450  -p 50051`
	- Install or otherwise have ardupilot SITL on your machine
	-  add `MCTSSA=33.246358,-117.425353,200,90` to your location.txt file located at `ardupilot/Tools/autotest/`
	- run the SITL using: `sim_vehicle.py -v ArduCopter --console --map --location MCTSSA --instance 1`
	- Manually give your drone a mission of fly-to points using the draw or edit features in MavProxy or otherwise get a mission to the drone
	- pull MavSDK-Java latest
	- Add [uploadIssueCode.txt](https://github.com/mavlink/MAVSDK/files/13923609/uploadIssueCode.txt) in `examples/java-client/src/main/java/io/mavsdk/example/` as a .java file
	- Run the file changing IS_RAW flag to true or false to change plugin used

I get times out on the request very quickly (1-2 seconds) with no log messages in Mavsdk.

SITL displays 
```
Got MISSION_ACK: TYPE_MISSION: OPERATION_CANCELLED
AP: Mission upload timeout:
```
sometime later about 10+ seconds after request

My console log below showing that position updates and mission download both work but upload does not:
```
10:44:38 AM: Executing ':MissionUploadIssue.main()'...

> Task :compileJava
> Task :processResources NO-SOURCE
> Task :classes

> Task :MissionUploadIssue.main()
10:44:39.152 [main] DEBUG io.mavsdk.example.MissionUploadIssue - Starting example: mission upload...
10:44:39.185 [main] DEBUG io.mavsdk.example.MissionUploadIssue - Creating RAW Mission Items...
10:44:39.185 [main] INFO io.mavsdk.example.MissionUploadIssue - Test MissionRaw Created!
10:44:39.185 [main] INFO io.mavsdk.example.MissionUploadIssue - ### START Raw Created Mission List ###
10:44:39.204 [main] INFO io.mavsdk.example.MissionUploadIssue - Mission Raw Item List:
MissionItem[0]: {seq = 0, frame = 0, command = 16, current = 1, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332463580, y = -1174253530, z = 200.09999, missionType = 0}, MissionItem[1]: {seq = 1, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332473225, y = -1174241931, z = 30.0, missionType = 0}, MissionItem[2]: {seq = 2, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332473436, y = -1174220750, z = 30.0, missionType = 0}, MissionItem[3]: {seq = 3, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332458991, y = -1174214572, z = 30.0, missionType = 0}, MissionItem[4]: {seq = 4, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332451715, y = -1174230962, z = 30.0, missionType = 0}, MissionItem[5]: {seq = 5, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332458253, y = -1174263995, z = 30.0, missionType = 0}, MissionItem[6]: {seq = 6, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332462365, y = -1174266768, z = 30.0, missionType = 0}, MissionItem[7]: {seq = 7, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332462365, y = -1174266768, z = 30.0, missionType = 0}, 
10:44:39.206 [main] INFO io.mavsdk.example.MissionUploadIssue - ### END Raw Created Mission List ###
10:44:39.206 [main] DEBUG io.mavsdk.example.MissionUploadIssue - Creating Standard Mission Items...
10:44:39.208 [main] INFO io.mavsdk.example.MissionUploadIssue - Test MissionStandard Created!
10:44:39.208 [main] INFO io.mavsdk.example.MissionUploadIssue - ### START Created Standard Mission List ###
10:44:39.209 [main] INFO io.mavsdk.example.MissionUploadIssue - Mission Raw Item List:
MissionItem[0]: {seq = 0, frame = 0, command = 16, current = 1, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332463580, y = -1174253530, z = 200.09999, missionType = 0}, MissionItem[1]: {seq = 1, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332473225, y = -1174241931, z = 30.0, missionType = 0}, MissionItem[2]: {seq = 2, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332473436, y = -1174220750, z = 30.0, missionType = 0}, MissionItem[3]: {seq = 3, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332458991, y = -1174214572, z = 30.0, missionType = 0}, MissionItem[4]: {seq = 4, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332451715, y = -1174230962, z = 30.0, missionType = 0}, MissionItem[5]: {seq = 5, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332458253, y = -1174263995, z = 30.0, missionType = 0}, MissionItem[6]: {seq = 6, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332462365, y = -1174266768, z = 30.0, missionType = 0}, MissionItem[7]: {seq = 7, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332462365, y = -1174266768, z = 30.0, missionType = 0}, 
10:44:39.209 [main] INFO io.mavsdk.example.MissionUploadIssue - ### END Created Standard Mission List ###
10:44:39.209 [main] INFO io.mavsdk.example.MissionUploadIssue - Subscribing to Telemetry updates...
10:44:39.349 [main] INFO io.mavsdk.example.MissionUploadIssue - Subscription Complete...
10:44:39.349 [main] INFO io.mavsdk.example.MissionUploadIssue - Downloading Mission...
10:44:39.592 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - ### START Downloaded Mission List ###
10:44:39.593 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - Mission Raw Item List:
MissionItem[0]: {seq = 0, frame = 0, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332463580, y = -1174253530, z = 200.09999, missionType = 0}, MissionItem[1]: {seq = 1, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332474174, y = -1174244201, z = 100.0, missionType = 0}, MissionItem[2]: {seq = 2, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332457093, y = -1174225919, z = 100.0, missionType = 0}, MissionItem[3]: {seq = 3, frame = 3, command = 16, current = 0, autocontinue = 1, param1 = 0.0, param2 = 0.0, param3 = 0.0, param4 = 0.0, x = 332453297, y = -1174258573, z = 100.0, missionType = 0}, 
10:44:39.593 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - ### END Downloaded Mission List ###
10:44:39.593 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - Attempting Mission Upload
10:44:40.444 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - Current Postition = [33.2463581,33.2463581,-0.025]
10:44:41.441 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - Current Postition = [33.2463581,33.2463581,-0.025]
10:44:42.173 [grpc-default-executor-1] INFO io.mavsdk.example.MissionUploadIssue - Standard Upload Mission Failed! Reason = TIMEOUT: Timeout
```

A Note: The mission upload will fail if you do not make one of the mission items (possibly the first one) have a "current" value of 1

JonasVautherin, 2024-01-12:
> A Note: The mission upload will fail if you do not make one of the mission items (possibly the first one) have a "current" value of 1

Yes, there is an offset between how PX4 and Ardupilot interprete the mission. It should be fixed for Ardupilot in the mission implementation. I think it should not be too much work, but... somebody using Ardupilot should look into it :innocent:.

Would you be willing to have a look?

acofis, 2024-01-12:
> Would you be willing to have a look?

I am about to take a deeper dive into the c++ code for another implementation so I will see if I can try and figure out the issue then. 

I am new to this code base and rusty with C++ so it may take me a lot more time than someone else to figure out the issue lol.

If someone else who uses Ardupilot gets the time or inclination that would also be a great help

JonasVautherin, 2024-01-13:
Also have a look at this, maybe: https://github.com/mavlink/MAVSDK/issues/2058

julianoes, 2024-01-14:
> I think it should not be too much work, but... somebody using Ardupilot should look into it 😇.

I don't think that will magically be fixed. That's been like this for 10 years and I don't think it's trivial to fix it an also make it backwards compatible.

JonasVautherin, 2024-01-14:
> I don't think that will magically be fixed. That's been like this for 10 years and I don't think it's trivial to fix it an also make it backwards compatible.

I was not referring to fixing this in MAVLink (I actually think that it is *not* possible in a backward compatible way). I was more thinking about some `if (ardupilot) { ... } else { ... }` in the MAVSDK mission logic. Don't you think that should be feasible?

julianoes, 2024-01-14:
Ah, yes, that's what we should do.

JonasVautherin, 2024-01-16:
> and the mission raw upload succeeds!

Oh you are using mission_raw? I always assumed it was mission, which has the indexing problem. The mission_raw plugin should work with Ardupilot, I am pretty sure that some people are using it already :+1:.

>  I plan to run a test to see exactly what the _sender.autopilot() is at that time but I don't have the time for a rebuild at the moment.

Let us know what you find!

chehrlic, 2024-03-07:
I encountered the same issue when using mission_raw. The problem is within
` Autopilot ServerComponentImpl::OurSender::autopilot() const
 {
     // FIXME: hard-coded to PX4 for now to avoid the dependency into mavsdk_impl.
    return Autopilot::Px4;
 }`
hard-coding `return Autopilot::ArduPilot;` fixes the issue for me but that's for sure not the correct solution :)

julianoes, 2024-03-07:
Hm, so you're creating a "fake" autopilot but want it to behave like ArduPilot, is that right?

chehrlic, 2024-03-08:
I don't think so - I'm just uploading waypoints to an ardupilot with mission_raw

julianoes, 2024-03-10:
But this todo is only relevant for components which are used for `_server` plugins. So, if you're just connecting to ArduPilot, you would not be using a server plugin.

Can you reproduce this with the latest version? And if so, what's the exact error?

chehrlic, 2024-03-10:
Ok, I did not use the really last version - looks like you fixed it in 30d2693417fe2afdfe494ace3e0f5b8ee5331f03 some weeks ago. But have to try it out during the next week an will report back if it really fixes the issue but it looks promising.

chehrlic, 2024-03-11:
I can confirm that 30d2693417fe2afdfe494ace3e0f5b8ee5331f03 fixes the timeout problem for me.

closed 2024-03-11 by julianoes

Sefibrah, 2024-06-24:
I think I have [the same issue](https://github.com/mavlink/MAVSDK-Java/issues/166), and it seems that you made it go away! What did you do? Please help if you can @chehrlic, I would appreciate it a ton!