mavlink/MAVSDK #2216: followme not found in C++

larcenn00, 2024-02-08:
hello, I can't get the example program follow_me.cpp to work.  In the console it displays the messages to me, with the coordinates generated in squares, but the drone remains at a static altitude of 2.5m.  
I'm on Linux Ubuntu 22.04 with simulator as recommended in the mavSDK documentation.  

The other examples work. 

 Is there anything special to do that I'm forgetting? 

Thanks a lot

julianoes, 2024-02-08:
What PX4 version is that?

larcenn00, 2024-02-09:
I use PX4-GAZEBO headless 1.14.0, i tested with 1.11.0 but same problem.

julianoes, 2024-02-10:
Ok, I will have to reproduce this. Basically, the follow me example doesn't do anything when you try it?

What's the output of the example when you run it?

larcenn00, 2024-02-10:
I supose to Followme example is correctly execute, i have this message : 

```
/MAVSDK/examples/follow_me$ build/follow_me udp://:14540
[02:20:24|Info ] MAVSDK version: v2.1.0 (mavsdk_impl.cpp:26)
[02:20:24|Info ] New system on: 172.17.0.2:14580 (with system ID: 1) (udp_connection.cpp:199)
[02:20:24|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:737)
[02:20:24|Debug] Component Autopilot (1) added. (system_impl.cpp:388)
[02:20:25|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:230)
[02:20:25|Debug] Discovered 1 component(s) (system_impl.cpp:563)
Waiting for system to be ready
System is ready
[02:20:26|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:256)
Armed
In Air...
[02:20:26|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m	 (system_impl.cpp:256)
[FlightMode: Takeoff] Target is at: nan, nan degrees.
Vehicle is at: 47.3978, 8.54561 degrees
[02:20:28|Debug] MAVLink: info: Takeoff detected	 (system_impl.cpp:256)
[FlightMode: Takeoff] Target is at: nan, nan degrees.
Vehicle is at: 47.3978, 8.54561 degrees
... (etc)
[02:20:31|Debug] Waiting for the system confirmation of the new configuration.. (follow_me_impl.cpp:119)
[FlightMode: Follow Me] Target is at: nan, nan degrees.
Vehicle is at: 47.3978, 8.54561 degrees
[FlightMode: Follow Me] Target is at: 47.3977, 8.54559 degrees.
Vehicle is at: 47.3978, 8.54561 degrees
[FlightMode: Follow Me] Target is at: 47.3974, 8.5457 degrees.
Vehicle is at: 47.3978, 8.54561 degrees
... (etc)
[02:21:12|Debug] MAVLink: info: Landing at current position	 (system_impl.cpp:256)
waiting until landed
Vehicle is at: 47.3978, 8.54561 degrees
waiting until landed
[02:21:13|Debug] MAVLink: info: Landing detected	 (system_impl.cpp:256)
Vehicle is at: 47.3978, 8.54561 degrees
Landed...
waiting until disarmed
Vehicle is at: 47.3978, 8.54561 degrees
waiting until disarmed
[02:21:16|Debug] MAVLink: info: Disarmed by landing	 (system_impl.cpp:256)
Vehicle is at: 47.3978, 8.54561 degrees
... (etc)
```

In the simulator prompt i have that : 

```
INFO  [commander] Ready for takeoff!
INFO  [commander] Armed by external command	
INFO  [tone_alarm] arming warning
INFO  [navigator] Using minimum takeoff altitude: 2.50 m	
INFO  [commander] Takeoff detected	
INFO  [commander] Landing at current position	
INFO  [commander] Landing detected	
INFO  [commander] Disarmed by landing	
INFO  [tone_alarm] notify neutral
INFO  [logger] closed logfile, bytes written: 16399233
```
In Qgroundcontrol the UAV execute the takeoff to 1.5m altitude, and down't move, and LAND.
In Qgroundcontrol during flight, the flightmode is "Followme"

julianoes, 2024-02-13:
I can reproduce this. It clearly doesn't work anymore.

@junwoo091400 you touched this last? Any chance you could have a quick check why the example doesn't work?

larcenn00, 2024-02-20:
@junwoo091400 i need help :P

larcenn00, 2024-06-29:
Hi, Obviously there has been no modification, is there a correction planned for this modification?

KarthiAru, 2024-08-01:
It works for me after I add this `target_location.absolute_altitude_m = 488.0;` to the example.
Video: https://youtu.be/vxwMRQwhBoc

I figured out that the altitude value of NaN is being sent in the mavlink message since there appears to be no default value set https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/follow_me/follow_me_impl.cpp#L320-L322

KarthiAru, 2024-08-01:
@julianoes Please find the PR - https://github.com/mavlink/MAVSDK/pull/2364

closed 2024-08-02 by julianoes