mavlink/MAVSDK #2220: Don't send rc_channels_override message

AnatolyAG, 2024-02-14:
Hi there.
I use MAVSDK version: v2.1.0

I want to emulate controlling the drone as if through a joystick. On python I used vehicle.channels.overrides['3'] = 2000 to control the elevation, and overwriting other channels. 
vehicle.channels.overrides['1'] = 1500
vehicle.channels.overrides['2'] = 1500
vehicle.channels.overrides['4'] = 1500
for flight control.
Now I am trying to do the same thing using c++.
I made a function that sends messages

```
void send_rc_channels_override(
    mavsdk::MavlinkPassthrough &mavlink_passthrough,
    uint8_t channel,
    uint16_t value)
{
    mavlink_passthrough.queue_message([&](MavlinkAddress mavlink_address, uint8_t channel)
        {
            mavlink_message_t message;
    uint16_t chan1_raw = 1;
    uint16_t chan2_raw = 1;  
    uint16_t chan3_raw = 1; 
    uint16_t chan4_raw = 1;  
    uint16_t chan5_raw = 1;  
    uint16_t chan6_raw = 1;  
    uint16_t chan7_raw = 1;  
    uint16_t chan8_raw = 1;  
    uint16_t chan9_raw = 1;  
    uint16_t chan10_raw = 0 ; 
    uint16_t chan11_raw = 0;  
    uint16_t chan12_raw = 0;  
    uint16_t chan13_raw = 0;  
    uint16_t chan14_raw = 0;  
    uint16_t chan15_raw = 0;  
    uint16_t chan16_raw = 0;  
    uint16_t chan17_raw = 0;  
    uint16_t chan18_raw = 0;

    mavlink_msg_rc_channels_override_pack_chan(
        mavlink_address.system_id, //uint8_t system_id,
        mavlink_address.component_id,//uint8_t component_id,
        0,  //uint8_t chan,
        &message, //mavlink_message_t * msg,
        mavlink_passthrough.get_target_sysid(), //uint8_t target_system, 
        mavlink_passthrough.get_target_compid(), // target_component, 
        chan1_raw, chan2_raw, chan3_raw, chan4_raw, chan5_raw, chan6_raw, chan7_raw, chan8_raw, chan9_raw,
        chan10_raw, chan11_raw, chan12_raw, chan13_raw, chan14_raw, chan15_raw, chan16_raw, chan17_raw, chan18_raw);

    std::cout << "channel - " << channel << " id - " << (int)message.msgid << std::endl;

    return message;
 });
   
}
```

 I called this function 

```
while (std::abs(dron_alt_curr - dron_alt_set) > 0.01)
    { //
        std::cout << "Try fly\n";
        send_rc_channels_override(mavlink_passthrough, 3, 2000);
        sleep_for(milliseconds(100));
    }
```

and using the sniffer example I subscribe to the messages I want on the other port.

```
mavsdk.intercept_incoming_messages_async([](mavlink_message_t& message) {
        if ((int)message.msgid == 0)
            return true;
        if (message.msgid != MAVLINK_MSG_ID_RC_CHANNELS_OVERRIDE)
        {
            return true;
        }
        
        std::cout << "Got message " << (int)message.msgid << '\n';
      
        std::cout << "  System ID: " << (int)message.sysid << std::endl;
        std::cout << "  Component ID: " << (int)message.compid << std::endl;
        std::cout << "  Sequence: " << (int)message.seq << std::endl;
        std::cout << "  Message length: " << (int)message.len << std::endl;
        // std::cout << "  Message name: " << messagename << std::endl;
        std::cout << "  Payload:" << std::endl;
        for (int i = 0; i < message.len; ++i) {
            std::cout << "    Byte " << i << ": " << (int)message.payload64[i] << std::endl;
        }
        return true;
    });
```

 But the emulator does not react to my messages and after a couple of seconds goes from armed to disarmed state. Maybe I'm doing something wrong, maybe there are examples of how to do it correctly? Or is there another way for solve my problem?

julianoes, 2024-02-15:
One nitpick:

```
mavlink_msg_rc_channels_override_pack_chan(
        mavlink_address.system_id, //uint8_t system_id,
        mavlink_address.component_id,//uint8_t component_id,
        0,  //uint8_t chan,
        &message, //mavlink_message_t * msg,
```

`0` for `chan` should be channel from the function arguments.

AnatolyAG, 2024-02-16:
Yes, I use channel from argument, I was the one who tried to go through the channels manually.

I usually do this

```
void send_rc_channels_override(
    mavsdk::MavlinkPassthrough &mavlink_passthrough,
    uint8_t channel,
    uint16_t value)
{

    mavlink_passthrough.queue_message([&](MavlinkAddress mavlink_address, uint8_t channel)
                                      {
            mavlink_message_t message;
    uint16_t chan1_raw = 1;
    uint16_t chan2_raw = 1;  
    uint16_t chan3_raw = 1; 
    uint16_t chan4_raw = 1;  
    uint16_t chan5_raw = 1;  
    uint16_t chan6_raw = 1;  
    uint16_t chan7_raw = 1;  
    uint16_t chan8_raw = 1;  
    uint16_t chan9_raw = 1;  
    uint16_t chan10_raw = 0 ; 
    uint16_t chan11_raw = 0;  
    uint16_t chan12_raw = 0;  
    uint16_t chan13_raw = 0;  
    uint16_t chan14_raw = 0;  
    uint16_t chan15_raw = 0;  
    uint16_t chan16_raw = 0;  
    uint16_t chan17_raw = 0;  
    uint16_t chan18_raw = 0;

    mavlink_msg_rc_channels_override_pack_chan(
        mavlink_address.system_id, //uint8_t system_id,
        mavlink_address.component_id,//uint8_t component_id,
        channel,  //uint8_t chan,
        &message, //mavlink_message_t * msg,
        mavlink_passthrough.get_target_sysid(), //uint8_t target_system, 
        mavlink_passthrough.get_target_compid(), // target_component, 
        chan1_raw, chan2_raw, chan3_raw, chan4_raw, chan5_raw, chan6_raw, chan7_raw, chan8_raw, chan9_raw,
        chan10_raw, chan11_raw, chan12_raw, chan13_raw, chan14_raw, chan15_raw, chan16_raw, chan17_raw, chan18_raw);

    std::cout << "channel - " << channel << " id - " << (int)message.msgid << std::endl;

    return message;
 });

  
}

```

and I subscribe  message 

```
mavsdk.intercept_incoming_messages_async([](mavlink_message_t& message) {
        if ((int)message.msgid == 0)
            return true;
        if (message.msgid != MAVLINK_MSG_ID_RC_CHANNELS_OVERRIDE)
        {
           return true;
        }
        
        std::cout << "Got message " << (int)message.msgid << '\n';

       std::cout << "  System ID: " << (int)message.sysid << std::endl;
        std::cout << "  Component ID: " << (int)message.compid << std::endl;
        std::cout << "  Sequence: " << (int)message.seq << std::endl;
        std::cout << "  Message length: " << (int)message.len << std::endl;
[... 209 lines cut ...]
    Byte 20: 0                    Byte 21: 0                    Byte 22: 0                   Byte 23: 0    Byte 24: 0
Got message 4
  System ID: 1
  Component ID: 195
  Sequence: 3
  Message length: 4
  Payload:
    Byte 0: 1785050240    Byte 1: 0    Byte 2: 1448036447    Byte 3: 9
Got message 4
  System ID: 1
  Component ID: 195
  Sequence: 4
  Message length: 4
  Payload:
    Byte 0: 1785102422    Byte 1: 0    Byte 2: 1448036447    Byte 3: 9
Got message 66
  System ID: 255
  Component ID: 190
  Sequence: 211
  Message length: 6
  Payload:
    Byte 0: -1091239934    Byte 1: 3    Byte 2: 1448036447    Byte 3: 9    Byte 4: 0    Byte 5: 0
```

maybe there are other ways?

julianoes, 2024-02-19:
So, the messages are arriving but PX4 doesn't use them? Or ArduPilot?

AnatolyAG, 2024-02-20:
So apparently it doesn't work for  the ArduPilot.

I use Mission planer - this is output when start

```
Setting SIM_SPEEDUP=1.000000
Suggested EK3_DRAG_BCOEF_* = 16.288, EK3_DRAG_MCOEF = 0.209
Home: 50.450725 30.461211 alt=160.800000m hdg=0.000000
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Loaded defaults from ....\Mission Planner\sitl\default_params/copter.parm
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
Smoothing reset at 0.001
validate_structures:498: Validating structures
Loaded defaults from .....\Mission Planner\sitl\default_params/copter.parm
New connection on serial port 0

```
I connect to 5762 port - control script
and connect to 5763 port - sniffer script.

I can't use this library for this case?

julianoes, 2024-02-22:
Are the messages arriving, yes or no?

AnatolyAG, 2024-02-22:
Yeah, apparently the message goes out and comes in, but the message code is wrong.
I may have phrased the question incorrectly.

AnatolyAG, 2024-02-23:
Thanks. I realized that the library works in a peculiar way with ArduPilot.

closed 2024-02-23 by AnatolyAG

JonasVautherin, 2024-02-23:
What do you mean by peculiar? Is there something to fix in MAVSDK? Can you elaborate a bit, in case someone else has the same issue?

AnatolyAG, 2024-02-24:
Don't mind me, it's a translation problem.
Let me describe again what I wanted to do and what I got.

I wanted to programmatically control the drone like an operator controls FPV, I see it this way, I write code that sends messages to the autopilot drone as if the operator from the remote control.
For experiments I chose mission planer, in it you can run emulation of different drones, I chose quadrocopter. Accordingly, it can be controlled by rewriting the control channels (which in my understanding emulates a joystick or remote control). 
I wrote the code in python using the dronekit library, in AltHold mode.
in this way `vehicle.channels.overrides['3'] = 2000` - I got the drone to the desired altitude, as the drone climbed I checked the current altitude from telemetry and upon reaching it using the channels 
```vehicle.channels.overrides['4'] - yaw,  
vehicle.channels.overrides['2'] # Pitch control.
vehicle.channels.overrides['1'] 0 # roll control.
```
I controlled the flight of the drone, as a result it is possible to emulate a fast flight and react to the current environment if you read and analyze the data from the sensors.
Then I tried using MAVSDK library to repeat the same actions and failed, for experiments I used the examples that come with the library, with a small modification.
From the noticed confusions :
1. Never once worked, everywhere disabled
```
 while (telemetry.health_all_ok() != true) {
    	std::cout << "Vehicle is getting ready to arm\n";
    	sleep_for(seconds(1));
	}
```	
2. Attempts to send a message to overwrite the channel as described above failed.
3. Using Offboard Mod - only managed to take off and at the same time it is changing the control mode.
4. Using Manual Control mode - failed to launch even in the example, right after the start there is a change of mode and the drone goes into disarm mode.

Here is a short what was the task and how it ended. If you can suggest something, I will be grateful.
If something I did wrong - I apologize, I'm just learning well and apparently how to do it correctly did not find, so do not kick too much.