mavlink/MAVSDK #2221: Discrepancy between Documentation and Example

BumbleIV, 2024-02-14:
I have the following basic `CMakeLists.txt` for my C++ code that makes use of the Telemetry API.

<details>
  <summary>CMakeLists.txt</summary>

```cpp

cmake_minimum_required(VERSION 3.15)
project(telemetry-stream)

# Set the C++ standard
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# Enable strict handling of warnings
add_definitions("-Wall -Wextra -Werror")

# Find MAVSDK package
find_package(MAVSDK REQUIRED)

# Add executable target with source files
add_executable(telemetry-stream main.cpp)

# Link MAVSDK library
target_link_libraries(telemetry-stream
  MAVSDK::mavsdk
)
```

</details>

<details>
  <summary>Output</summary>

```
-- Configuring done (0.1s) CMake Error at CMakeLists.txt:18 (target_link_libraries): Target "telemetry-stream" links to:
MAVSDK::mavsdk_telemetry
but the target was not found. Possible reasons include:

* There is a typo in the target name.
* A find_package call is missing for an IMPORTED target.
* An ALIAS target is missing.
-- Generating done (0.0s)
CMake Generate step failed. Build files cannot be regenerated correctly.
```

</details>

According to [this documentation](https://mavsdk.mavlink.io/main/en/cpp/guide/telemetry.html), `MAVSDK::mavsdk_telemetry` is needed within the `target_link_libraries` call.

However, this [following example's](https://mavsdk.mavlink.io/main/en/cpp/examples/takeoff_and_land.html) source code which makes use of Telemetry does **_NOT_** include `MAVSDK::mavsdk_telemetry` within the `target_link_libraries` call.

If I delete `MAVSDK::mavsdk_telemetry`, I am able to build and execute. 

However, I wanted to ask: why there is a discrepancy between the documentation and the example, and which I should follow?

JonasVautherin, 2024-02-15:
With recent versions of MAVSDK, you only need `MAVSDK::mavsdk`. It appears that the documentation is outdated here.

Would you be willing to open a PR and correct the documentation? That would be highly appreciated :upside_down_face:.

BumbleIV, 2024-02-15:
I would love to! Will do :)

julianoes, 2024-02-15:
It's here:
https://github.com/mavlink/MAVSDK-docs/blob/main/en/cpp/guide/telemetry.md?plain=1#L54-L65

KarthiAru, 2024-07-29:
Created a PR - https://github.com/mavlink/MAVSDK-docs/pull/287

julianoes, 2024-07-29:
Merged, thanks @KarthiAru, and closing this.

closed 2024-07-29 by julianoes