mavlink/MAVSDK #2225: Installation with Cmake: Doesn't build with python3.12 or higher: ModuleNotFoundError: No module named 'imp'

pawsaw, 2024-02-17:
### Problem description

In my projects CMakeLists.txt:

```
include(FetchContent)
FetchContent_Declare(
  mavsdk
  GIT_REPOSITORY https://github.com/mavlink/MAVSDK.git
  GIT_TAG        v2.2.0
  GIT_SUBMODULES_RECURSE TRUE
)
FetchContent_MakeAvailable(mavsdk)
```

If cmake picks Python 3.12 (or higher) the build fails:
```[cmake] ModuleNotFoundError: No module named 'imp'```

**imp has been removed in Python 3.12**

### Expected behavior

Works with Python >= 3.12**

### Side note:
I tried with 
```
unset(Python3_EXECUTABLE)
find_package (Python 3.10 EXACT REQUIRED COMPONENTS Interpreter Development)
``` 
in cmake. However, this seems not to apply to dependencies as fetched by FetchContent_*

**Idea**
1. Maybe adding 

```
unset(Python3_EXECUTABLE)
find_package (Python 3.10 EXACT REQUIRED COMPONENTS Interpreter Development)
``` 
to the MavSDK CMakeLists.txt fixes the issue?

2. Get rid of *imp*

julianoes, 2024-02-19:
We should probably do 2. Do you know where it's required?

leoplo, 2024-03-27:
This was happening in `future` egg but was corrected in `18.4` release: https://github.com/PythonCharmers/python-future/pull/632 .

julianoes, 2024-03-27:
Ok, so potentially with latest `future` it should be ok again?

leoplo, 2024-03-28:
On my side at least it solved the problem.

julianoes, 2024-03-28:
Nice, ok!

closed 2024-05-27 by julianoes