mavlink/MAVSDK #2228: problem in building mavsdk exmaple on raspbian os

AvanthikaSelva, 2024-02-22:
when i  am trying to run the mavsdk examples by refering the mavsdk documentation . At first step i tried building it 
 step  1) cd examples/takeoff_and_land/ ...this command got runned 
step  2) cmake -Bbuild -H. .........while running this command  i got the following 

usage
       cmake [options] <path-to-source>
       cmake [options] <path-to-existing-build>
       cmake [options] -S <path-to-source> -B <path-to-build>
specify a source directory to (re-)generate a build system for it in the current working directory .  Specify an existing build directory to re-generate its builld system.

          help me with this issue

JonasVautherin, 2024-02-22:
On what system are you running? Can you try `cmake -Bbuild -S.` in step 2, instead of `-H.`?

julianoes, 2024-02-22:
What's your cmake version?

julianoes, 2024-05-21:
No response, closing.

closed 2024-05-21 by julianoes