mavlink/MAVSDK #2248: Building MAVSDK for Android failing (Apple Silicon M2 Max)

rayw-dronesense, 2024-03-13:
## Environment

- Apple M2 Max Macbook Pro

- 64 GB RAM

- Mac OS Ventura

- Docker Desktop 4.28.0

## Problem

I cloned the MAVSDK repo on the main branch, and tried to run the configuration step following the documentation to use dockcross.

However, it fails every single time when it tries to build OpenSSL,

```
make[3]: *** write jobserver: Bad file descriptor.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[3]: *** write jobserver: Bad file descriptor.  Stop.
gmake[2]: *** [CMakeFiles/openssl.dir/build.make:86: openssl/src/openssl-stamp/openssl-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/openssl.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  /work/build/default/third_party/openssl failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:23 (build_target)

-- Configuring incomplete, errors occurred!
```

The command being used to configure the build is

```
~/dockcross-android-arm64 cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=ON -H.
```

The end goal is to make modifications to MAVSDK C++ and bring it into the Android projects (mavsdk_server.aar and mavsdk-java.jar)

JonasVautherin, 2024-03-13:
It is interesting that the build happening inside a docker container fails on the M2 :thinking:.

Do you know why that would be? What is this "Bad file descriptor"?

Jai-GAY, 2024-03-14:
> The end goal is to make modifications to MAVSDK C++ and bring it into the Android projects (mavsdk_server.aar and mavsdk-java.jar)

What is the [difference between these two](https://github.com/mavlink/MAVSDK-Java)? I hope I can further understand and learn.

rayw-dronesense, 2024-03-14:
> It is interesting that the build happening inside a docker container fails on the M2 🤔.
> 
> Do you know why that would be? What is this "Bad file descriptor"?

I have no idea. Hence, reaching out to the experts for help 😁

rayw-dronesense, 2024-03-14:
> What is the [difference between these two](https://github.com/mavlink/MAVSDK-Java)? I hope I can further understand and learn.

@Jai-GAY

 
MAVSDK-Java is a wrapper interface that talks to the 'MAVSDK-server' using generated protobuf interfaces. The 'Server' contains a pre-built .so binary of MAVSDK C++, in this case, targets Android.

Just having the Java wrapper and Server isn't enough, if you are looking to make modifications because the meat underneath is actually within the MAVSDK C++ project.

JonasVautherin, 2024-03-14:
It seems like openssl fails to build with dockcross on the M2. I would try to remove MAVSDK from the equation and just build OpenSSL first. Then see if that fails or not, and go from there.

[This is where we build openssl](https://github.com/mavlink/MAVSDK/tree/main/third_party/openssl), you could start from there.

Jai-GAY, 2024-03-15:
> if you are looking to make modifications because the meat underneath is actually within the MAVSDK C++ project.

okay, so as an open-source spirit, isn't it a pull request, submit?

julianoes, 2024-04-11:
FYI: https://github.com/openssl/openssl/issues/23845

julianoes, 2024-04-11:
@rayw-dronesense are you still looking into this?

julianoes, 2024-05-27:
OP disappeared, not sure what I should do, closing.

closed 2024-05-27 by julianoes