mavlink/MAVSDK #2255: Source file cmake gtest cannot find cmakelist.txt

YuZhe-CN, 2024-03-23:
Compiling the source files and installing the library with it, cmake shows an error about not finding cmakelist.txt in gtest.
[image]
But cloning the repository is all fine

JonasVautherin, 2024-03-23:
It really looks like the submodule was not cloned properly. Though I see that you run `git submodule update --init --recursive` :thinking:. Can you check what is in `src/third_party/gtest` after that?

Also you can do it in one line:

```
RUN git clone https://github.com/mavlink/MAVSDK.git --recursive
```

YuZhe-CN, 2024-03-23:
In the beginning, I was trying to build the library with the source code on the release. After the error showed up, I checked the contents inside the gtest folder. And it is empty. 
[image]
After Trying cloning the git repository directly and updating the submodule, the result is shown below. 
[image]

JonasVautherin, 2024-03-24:
Awesome!

closed 2024-03-24 by JonasVautherin