mavlink/MAVSDK #2259: Is MAVSDK able to accept multiple drones connections from one UDP port? If it is posible, is add_udp_connection blocking or non blocking?

YuZhe-CN, 2024-03-27:
Hi support, 

I'm trying to write a server that accepts multiple connections from one UDP port. My idea is to let a thread always waiting for the connections. But, I don't know if it is possible. 

Thank you

julianoes, 2024-03-27:
Can you give some context of what you're trying to do overall without the implementation specifics?

YuZhe-CN, 2024-03-27:
Well, I want to implement a server that is awaiting the connections of drones. This server is listening on one port. The main function of the server is to give orders like go to the point (x, y) or follow a predefined route and meanwhile, the drones will transfer images or videos to another server that will process all the data. This second server's IP address and Port information is passed to the drone when it connects to the first server. 

To sum up, there will be two servers, one that only gives orders and missions to the drones and check their states. And another that will process the photos.

julianoes, 2024-03-27:
It sounds like to me that you need to:
1. Listen on a default port, e.g. 14550 for drones to appear.
2. Once a drone is connected, you would reconfigure it to another port using some sort of mechanism that you have to implement
3. Then the drone will switch to that other port, leaving 14550 free again for the next one.

It would basically be some sort of discovery mechanism on UDP.

You could read this to learn how zeromq discovery does it:
https://zguide.zeromq.org/docs/chapter8/

julianoes, 2024-03-27:
I'm going to close this as this is clearly just a question and not a MAVSDK issue as such.

closed 2024-03-27 by julianoes