mavlink/MAVSDK #2262: Unable to subscribe to Velocity Body Fields

Swabhan, 2024-03-28:
I am currently trying to get the telemetry data that represents velocity in the x, y, and z direction. In the documentation I have found this portion that should allow me to subscribe and get the fields, but unfortunately I am unable to do it as easily as others. 

Link to Documentation
https://mavsdk.mavlink.io/main/en/cpp/api_reference/structmavsdk_1_1_telemetry_1_1_velocity_body.html

I have tried subscribe_velocity_ned, subscribe_velocity_body, and more with no success.

julianoes, 2024-03-28:
@Swabhan have you tried to set the rate to make sure it's sent out?
https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_telemetry.html#classmavsdk_1_1_telemetry_1ab5cb79fd53f27f245808a6bb9ed3225d

Is this with PX4 or ArduPilot?

Swabhan, 2024-03-28:
I am using a virtual drone with JMAV and PX4, why would I need to set the rate, wouldn't I just need to subscribe to get continuous data flow from my ground controller?

`telemetry->subscribe_attitude_angular_velocity_body([](Telemetry::AngularVelocityBody angularVelocity) {
        updateTelemetryData({
            {"roll_rad_s", angularVelocity.roll_rad_s},
            {"pitch_rad_s", angularVelocity.pitch_rad_s},
            {"yaw_rad_s", angularVelocity.yaw_rad_s},
        });
    });

    telemetry->subscribe_velocity_ned([](Telemetry::VelocityNed velocity) {
        updateTelemetryData({
            {"north_m_s", velocity.north_m_s},
            {"east_m_s", velocity.east_m_s},
            {"down_m_s", velocity.down_m_s}
        });
    });

    telemetry->subscribe_fixedwing_metrics([](Telemetry::FixedwingMetrics metrics) {
        updateTelemetryData({
            {"airspeed_m_s", metrics.airspeed_m_s},
            {"throttle_percentage", metrics.throttle_percentage},
            {"climb_rate_m_s", metrics.climb_rate_m_s}
        });
    });`

the above code works fine.

Thank you for the quick reply

julianoes, 2024-03-28:
Ok, in SITL it will indeed already stream it. So, I missed what you changed to make it work?