mavlink/MAVSDK #2263: Can I use mavsdk::MavlinkPassthrough::queue_message() to transmit the GPS_GLOBAL_ORIGIN message to PX4 to specify the ekf origin?

747640013, 2024-03-28:
```
auto now = std::chrono::system_clock::now();

  auto microseconds = std::chrono::duration_cast<std::chrono::microseconds>(now.time_since_epoch()).count();

    const MavlinkPassthrough::Result pass_result = mavlink_passthrough.queue_message([microseconds](MavlinkAddress mavlink_address,uint8_t channel)->mavlink_message_t{
        mavlink_message_t msg;
        mavlink_address.system_id=1;
        mavlink_address.component_id=1;
        mavlink_msg_gps_global_origin_pack_chan(
            mavlink_address.system_id,
            mavlink_address.component_id,
            MAVLINK_COMM_0,
            &msg,
            41.15403,
            116.2593928,
            57.982,
            microseconds
        );
        return msg;
    });
    ```

julianoes, 2024-03-28:
I'm not sure this message is used in the way you think when using HITL.

Have a look at how GPS is sent in the Gazebo plugin.

747640013, 2024-03-29:
Thanks for your help, I wrote these lines of code for the takeoff and landing examples. My hardware connection method is to connect Raspberry Pi 4b and pixhawk through USB to TTL via telem port, and then connect pixhawk to my laptop through USB for HITL.
I'm going to learn how GPS is sent in the Gazebo plugin.

747640013, 2024-03-29:
I used the wrong message id

closed 2024-03-29 by 747640013