mavlink/MAVSDK #2266: MAVSDK_SERVER build error: 'undefined reference to `EVP_MAC_free''

rayhannarindran, 2024-03-30:
Have been trying to build mavsdk_server for a while, new to gcs development. I have looked around everywhere to find the fix, from OpenSSL versions, changing build flags, nothing worked. mavsdk_server did not show up

Currently on ubuntu 20.04, if anyone have a fix or anything i mightve done wrong please let me know thank you!

Command used
>cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/default -H.
cmake --build build/default -j8

Error
> /usr/bin/ld: ../../../third_party/install/lib/libgrpc.a(json_token.cc.o): in function `compute_and_encode_signature(grpc_auth_json_key const*, char const*, char const*)':
json_token.cc:(.text+0x66a): undefined reference to `EVP_DigestSignUpdate'
/usr/bin/ld: ../../../third_party/install/lib/libgrpc.a(ssl_transport_security.cc.o): in function `ssl_handshaker_result_extract_peer(tsi_handshaker_result const*, tsi_peer*)':
ssl_transport_security.cc:(.text+0x3096): undefined reference to `SSL_get1_peer_certificate'
/usr/bin/ld: ../../../third_party/install/lib/libgrpc.a(aws_request_signer.cc.o): in function `grpc_core::(anonymous namespace)::SHA256Hex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
aws_request_signer.cc:(.text+0x2c9): undefined reference to `EVP_Q_digest'
/usr/bin/ld: ../../../third_party/install/lib/libgrpc.a(aes_gcm.cc.o): in function `aes_gcm_derive_aead_key(absl::lts_20240116::Span<unsigned char>, unsigned char*, absl::lts_20240116::Span<unsigned char const>, absl::lts_20240116::Span<unsigned char const>)':
aes_gcm.cc:(.text+0xba): undefined reference to `EVP_MAC_fetch'
/usr/bin/ld: aes_gcm.cc:(.text+0xc5): undefined reference to `EVP_MAC_CTX_new'
/usr/bin/ld: aes_gcm.cc:(.text+0xef): undefined reference to `OSSL_PARAM_construct_utf8_string'
/usr/bin/ld: aes_gcm.cc:(.text+0x11b): undefined reference to `OSSL_PARAM_construct_end'
/usr/bin/ld: aes_gcm.cc:(.text+0x157): undefined reference to `EVP_MAC_init'
/usr/bin/ld: aes_gcm.cc:(.text+0x169): undefined reference to `EVP_MAC_update'
/usr/bin/ld: aes_gcm.cc:(.text+0x17f): undefined reference to `EVP_MAC_update'
/usr/bin/ld: aes_gcm.cc:(.text+0x19a): undefined reference to `EVP_MAC_final'
/usr/bin/ld: aes_gcm.cc:(.text+0x1a6): undefined reference to `EVP_MAC_CTX_free'
/usr/bin/ld: aes_gcm.cc:(.text+0x1ae): undefined reference to `EVP_MAC_free'
/usr/bin/ld: aes_gcm.cc:(.text+0x1d4): undefined reference to `EVP_MAC_CTX_free'
/usr/bin/ld: aes_gcm.cc:(.text+0x1dc): undefined reference to `EVP_MAC_free'
collect2: error: ld returned 1 exit status
make[2]: *** [src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/build.make:182: src/mavsdk_server/src/mavsdk_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:1326: src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

And all i have in the mavsdk_server/src directory is as such:
>rayhannarindran@NarindransUbuntu:~/programming/MAVSDK/build/default/src/mavsdk_server/src$ ls
CMakeFiles  cmake_install.cmake  CTestTestfile.cmake  libmavsdk_server.a  Makefile

rayhannarindran, 2024-03-30:
I have fixed it, turns out i needed to update cmake to the latest version, by removing with `sudo apt remove cmake` and installing it again weirdly with 'pip install cmake --upgrade'

closed 2024-03-30 by rayhannarindran

JonasVautherin, 2024-03-30:
Note that you don't have to install it from pip (I wouldn't do that). You can either download it from the cmake website, or if you use snaps (I don't) you can install it through there.