mavlink/MAVSDK #2273: Info plugin blocking on async command

piatf, 2024-04-05:
Hello !

I am working with an autopilot communicating with MAVSDK. 
In the constructor of my Autopilot component, I want to create an "Info" plugin (and some other ones).
Unfortunately, the program **blocks** upon my component construction.

After a first investigation, I can see that the following happens : 

1) Heartbeat is received in MAVSDK => component is discovered
2) Connection is being enabled, and plugins start enabling too.
3) Info plugin is enabling, and is waiting to receive autopilot version and flight information...
(let's consider only autopilot version at first)

_Extract from [info_impl.cpp](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/info/info_impl.cpp) :_
```
void InfoImpl::enable()
{
    // We can't rely on System to request the autopilot_version,
    // so we do it here, anyway.
    _system_impl->send_autopilot_version_request();
    _system_impl->send_flight_information_request();
...
...
...
void SystemImpl::send_autopilot_version_request()
{
    auto prom = std::promise<MavlinkCommandSender::Result>();
    auto fut = prom.get_future();

    send_autopilot_version_request_async(
        [&prom](MavlinkCommandSender::Result result, float) { prom.set_value(result); });

    if (fut.get() == MavlinkCommandSender::Result::Unsupported) {
        _old_message_520_supported = false;
        LogWarn() << "Trying alternative command (512).";
        send_autopilot_version_request();
    }
...
...
(same principle for send_flight_information_request)
```
4) The std::promise is waiting to receive and process an AUTOPILOT VERSION message,
BUT, 
by logging the thread ID, I understood that the thread that is processing HEARTBEAT, and enabling plugins is the same that is suppose to process any other messages, including AUTOPILOT VERSION.

**=> As a conclusion**, the async command is waiting to receive a message that cannot be processed, because the message reception is normally handled by the current thread, which is currently processing HEARTBEAT.

**Related information :** 
- I was not able to reproduce it with some examples provided.
- I run MAVSDK v2.6.0, but I encounter the issue since v2.0.0.
- I am using Ubuntu 22.04.
- I was able to find a workaround for my case, by passing `nullptr` to `send_autopilot_version_request_async`, and ignoring the handling of MAVLink msgId 520 (as mentionned in the code comments, MAVLINK is now recent enough to use REQUEST_MESSAGE instead). And I did the same thing for flight information request.

julianoes, 2024-04-06:
Thanks for the issue and the detailed information.

I think you're right that we're not meant to block in the `enable()` call. I wonder why we don't always run into this issue though? :thinking:

piatf, 2024-04-06:
I know, I wonder about it as well.
The only thing that I am sure, is that autopilot version is sent back from autopilot : I was able to check it with Wireshark, and all values seem ok.

julianoes, 2024-04-06:
Do you have an easy way for me to reproduce? E.g. with PX4 SITL?

piatf, 2024-04-08:
The code I am using is a bit complex and implies some other classes, but I was able to write a simple example that reproduce the issue: 

```
#include <mavsdk.h>
#include <mavsdk/plugins/info/info.h>
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <system.h>
#include <thread>

class MyComponent {
public:
  MyComponent(std::shared_ptr<mavsdk::System> system)
      : telemetry_(std::make_shared<mavsdk::Telemetry>(system)),
        // Blocking on info plugin enabling step
        info_(std::make_shared<mavsdk::Info>(system)) { 
    fprintf(stdout, "AUTOPILOT connected\n");
  }

private:
  std::shared_ptr<mavsdk::Telemetry> telemetry_;
  std::shared_ptr<mavsdk::Info> info_;
};

int main(int argc, char *argv[]) {
  (void)argc;
  (void)argv;

  auto my_mavsdk = std::make_shared<mavsdk::Mavsdk>(
      mavsdk::Mavsdk::Configuration{1, MAV_COMP_ID_ONBOARD_COMPUTER, true});
  my_mavsdk->add_any_connection("udp://127.0.0.1:14580");

  auto system = my_mavsdk->first_autopilot(3.0).value();

  my_mavsdk->subscribe_on_new_system([my_mavsdk, system]() {
    auto my_component = std::make_shared<MyComponent>(system);
    (void)my_component;
  });

  while (true) {
    std::this_thread::sleep_for(std::chrono::milliseconds(10));
  }
  return EXIT_SUCCESS;
}
```

This is a "quick and dirty" draft, but I reproduces the issue with it, and PX4 Autopilot v1.14 SITL.

julianoes, 2024-04-11:
Ok, thanks for the example.

The way you do the discovery seems a bit odd to me:

```
auto system = my_mavsdk->first_autopilot(3.0).value();
```

I assume you're aware that this will give you a `nullptr` if it times out and doesn't discover an autopilot.

```
my_mavsdk->subscribe_on_new_system([my_mavsdk, system]() {
    auto my_component = std::make_shared<MyComponent>(system);
    (void)my_component;
  });
```

Why do you use `subscribe_on_new_system` when you have already waited for an autopilot? It's usually either or, so either you just wait for the first autopilot, or you need more fine grained control and you use `subscribe_on_new_system` to discover any system as they connect.

Here are two example ways to do this discovery:

Using `first_autopilot()`:

https://github.com/mavlink/MAVSDK/blob/324b64543d5ec1ad3d480b6d8a13d6cdc67ee68b/examples/takeoff_and_land/takeoff_and_land.cpp#L44-L52

And using `subscribe_on_new_system()`:

https://github.com/mavlink/MAVSDK/blob/324b64543d5ec1ad3d480b6d8a13d6cdc67ee68b/examples/fly_multiple_drones/fly_multiple_drones.cpp#L104-L123

or:

https://github.com/mavlink/MAVSDK/blob/324b64543d5ec1ad3d480b6d8a13d6cdc67ee68b/examples/camera_zoom/camera_zoom.cpp#L47-L60

piatf, 2024-04-11:
Hello, 

Sorry for that, I may have tried to over-simplify the code, and I didn't see that little mistake.
I re-wrote the example thanks to your help, and here is the correct main() function : 

```cpp
.....
.....

int main(int argc, char *argv[]) {
  (void)argc;
  (void)argv;

  auto my_mavsdk = std::make_shared<mavsdk::Mavsdk>(
      mavsdk::Mavsdk::Configuration{1, MAV_COMP_ID_ONBOARD_COMPUTER, true});
  my_mavsdk->add_any_connection("udp://127.0.0.1:14580");

  std::cout << "Waiting to discover system...\n";

  my_mavsdk->subscribe_on_new_system([my_mavsdk]() {
    std::cout << "New system discovered, calling callback...\n";

    const auto system = my_mavsdk->systems().back();
    auto my_component = std::make_shared<MyComponent>(system);
    (void)my_component;

    std::cout << "New system discovered, callback ended...\n";
  });

  while (true) {
    std::this_thread::sleep_for(std::chrono::milliseconds(10));
  }
  return EXIT_SUCCESS;
}
```

I also logged HEARTBEAT reception, and added 3 logs in `send_autopilot_version_request`, to help debugging it :
```cpp
void SystemImpl::send_autopilot_version_request()
{
    auto prom = std::promise<MavlinkCommandSender::Result>();
    auto fut = prom.get_future();

    std::cout << "send_autopilot_version_request_async 1" << std::endl;
    send_autopilot_version_request_async(
        [&prom](MavlinkCommandSender::Result result, float) { prom.set_value(result); });

    std::cout << "send_autopilot_version_request_async 2" << std::endl;
    if (fut.get() == MavlinkCommandSender::Result::Unsupported) {
        _old_message_520_supported = false;
        LogWarn() << "Trying alternative command (512).";
        send_autopilot_version_request();
    }
    std::cout << "send_autopilot_version_request_async 3" << std::endl;
}
```

Considering above information, here is the ouput : 

```
[10:06:42|Info ] MAVSDK version: v1.4.0-765-gc9512b9f (mavsdk_impl.cpp:26)
[10:06:42|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:735)
Waiting to discover system...
[10:06:43|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:378)
[10:06:43|Debug] Component Autopilot (1) added. (system_impl.cpp:389)
New system discovered, calling callback...
AUTOPILOT connected
New system discovered, callback ended...
HEARTBEAT RECEIVED
[10:06:44|Debug] Discovered 1 component(s) (system_impl.cpp:567)
New system discovered, calling callback...
send_autopilot_version_request_async 1
send_autopilot_version_request_async 2
[10:06:45|Warn ] Callback took more than 1 second to run.
See: https://mavsdk.mavlink.io/main/en/cpp/troubleshooting.html#user_callbacks (mavsdk_impl.cpp:852)
HEARTBEAT RECEIVED
HEARTBEAT RECEIVED
HEARTBEAT RECEIVED
```

We can see that `send_autopilot_version_request_async 3` is never reached and program still blocks. 
The only thing that confuses me is the log `AUTOPILOT connected` which is visible, and I don't understand why.

PS : I won't be able to work on that topic ( and I will probably won't reply on this thread ) for the next couple of weeks.

julianoes, 2024-05-20:
I think I ran into this again recently and somewhat fixed it:
https://github.com/mavlink/MAVSDK/pull/2301/commits/9405585066997c630751132d41f36547b14a0f5d

closed 2024-05-20 by julianoes