mavlink/MAVSDK #2278: QtCreator adding librrary

TkaczykHub, 2024-04-11:
Can someone explain me how to add library to use in Qt Project App?

JonasVautherin, 2024-04-11:
I am sorry but this has nothing to do with MAVSDK, does it? :thinking: 

I suggest you look into the QtCreator documentation (maybe they have a forum or something, too?).

closed 2024-04-11 by JonasVautherin

TkaczykHub, 2024-04-11:
Is it impossible?

julianoes, 2024-04-11:
I'm sure it's possible. Mavsdk is like any other C++ dependency. Are you building your Qt project using qmake or cmake?

TkaczykHub, 2024-04-12:
yes, im using cmake and i need help to import all mavsdk to my files, cause i want to make app in Qt with MAVSDK possibiliets. can you help me to make "mavsdk.h" works?

julianoes, 2024-04-12:
If you're using cmake you should add it as any of the examples:
https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_and_land/CMakeLists.txt