mavlink/MAVSDK #2287: Getting HTTP 503 when attempting to subscribe to telemetry but works after connecting with QGroundControl

tltanlum, 2024-04-18:
When I connect with `mavsdk_server_bin` to my drone running `ArduPilot`, I cannot subscribe for telemetry as my web client is receiving 503 from `mavsdk_server_bin`. It is the same behavior with `APM Planner` and I connect through the serial port in both cases(mavsdk_server_bin and APM Planner) mutually exclusively.

Once I connect with `QGroundControl` through the serial port, all the data starts streaming. From there on I can connect with either `APM Planner` or `mavsdk_server_bin` and they both receive the data just fine; and my web client can subscribe to telemetry just fine through `mavsdk_server_bin`.

I am not sure what triggers the data to start streaming through QGroundControl and I would expect this to be the norm to be honest. I am wondering if there is an ArduPilot setting to have this behavior enabled by default for all clients connecting to mavlink through serial/tcp/udp?

If I hardware reset the ArduPilot the same problem comes up as I need to run QGroundControl which somehow triggers the mavlink streaming.

APM-Planner v2-2.0.30-rc3
QGrountControl v4.3.0
MAVSDK-Server v2.9.1

Thank you!

JonasVautherin, 2024-04-18:
> my web client

What web client is that? How do you connect to mavsdk_server_bin exactly?

julianoes, 2024-04-19:
Can you try to use the `set_rate_` commands to get messages flowing?

tltanlum, 2024-04-19:
> > my web client
> 
> What web client is that? How do you connect to mavsdk_server_bin exactly?

It is a typescript app using mavsdk-proto

tltanlum, 2024-04-19:
> Can you try to use the `set_rate_` commands to get messages flowing?

This was my suspicion as well but even after adding the `set_rate` calls, I am still getting the following response headers:

```
grpc-message
	upstream connect error or disconnect/reset before headers. reset reason: connection failure, transport failure reason: delayed connect error: 111
grpc-status
	14
```

In the `mavsdk_server_bin logs` I see fail-safe critical errors. Could this affect the telemetry subscription; I mean does the `mavsdk_server_bin` have a precondition of having all fail-safe mechanisms in place before allowing subscriptions?

```
Apr 18 11:45:15 mavsdk_server[200337]: [11:45:15|Info ] MAVSDK version: v2.7.0 (mavsdk_impl.cpp:26)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:15|Info ] Waiting to discover system on serial:///dev/MAVLINK_1:115200... (connection_initiator.h:20)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:16|Debug] New system ID: 1 Comp ID: 0 (mavsdk_impl.cpp:737)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:16|Debug] Component Autopilot (1) added. (system_impl.cpp:389)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:16|Warn ] Vehicle type changed (new type: 13, old type: 0) (system_impl.cpp:231)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:25|Debug] MAVLink: critical: PreArm: RC not found (system_impl.cpp:257)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:25|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:257)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:56|Debug] MAVLink: critical: PreArm: RC not found (system_impl.cpp:257)
Apr 18 11:46:58 mavsdk_server[200337]: [11:45:56|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:257)
```

tltanlum, 2024-04-19:
The key point is the `QGroundControl` enabling everything upon connection whereas `APM Planner 2` and `mavsdk_server_bin` not doing the same. I did not have this problem with the `px4-gazebo-headless` simulator both UIs receiving data but also `mavsdk_server_bin`. All worked with it upon connection.

I changed the telemetry rates from `APM Planner 2` and restarted the UI but still not getting anything

julianoes, 2024-04-20:
Ok, it looks like your error is more on the gRPC rather than the MAVLink side then. I don't see your typescript side, so I'm not sure what's wrong. All I know is that `mavsdk_server` generally works for other wrappers like Python...

tltanlum, 2024-04-22:
OK I forgot to mention that the serial port is propagated over TCP through wifi mostly because I did not expect it to be an issue since `QGroundControl` works. I connected the pilot directly on the machine running the `mavsdk_server` and it connects just fine on the real serial port, with minimum baud rate of `115200`. But it seems that QGroundControl somehow adjusts the serial connection regardless if local or over TCP and it works with any configuration. I will need to check the `QGroundControl` code to check what are the measures taken by the `LinkManager` component to make it work, regardless.

julianoes, 2024-04-23:
I'm afraid I'm completely confused. To me, it sounds like you're mixing up MAVLink and gRPC. 

> in the mavsdk_server_bin logs I see fail-safe critical errors. Could this affect the telemetry subscription; I mean does the mavsdk_server_bin have a precondition of having all fail-safe mechanisms in place before allowing subscriptions?

It looks like mavsdk_server has connected fine. The fact that you don't see telemetry updates, in my view, is entirely on **your** implementation of the gRPC side, so you gRPC client. You said this is your own typescript implementation, so I don't think I can help with something that I don't see.

tltanlum, 2024-04-24:
> I'm afraid I'm completely confused. To me, it sounds like you're mixing up MAVLink and gRPC.
>
>
> It looks like mavsdk_server has connected fine. The fact that you don't see telemetry updates, in my [...]

I set the telemetry rate parameter to 10Hz as default directly on the `ArduPilot` through `QGroundControl`; now `mavsdk_server_bin` does not return a 503 unavailable anymore upon sending `set_rate` through the typescript client and it is receiving the telemetry right upon connection and subscription to the telemetry. The typescript client was irrelevant because either `set_rate` or `subscribe` would always return `503`. Whatever the problem was is related to the virtual serial port which is propagated through wifi from another node. QGroundControl seems to be managing though in any case with its serial connection component

closed 2024-04-24 by tltanlum