mavlink/MAVSDK #2288: Why is takeoff called here? I saw that takeoff is set in the create_mission_raw function. For APM, do you need to set takeoff in the mission and also call takeoff outside? Why?

wenii, 2024-04-24:
https://github.com/mavlink/MAVSDK/blob/849cb9a3c3bd18d21731da3b0c4fe31ca6b9d039/src/integration_tests/mission_raw_import_and_fly.cpp#L127

julianoes, 2024-04-26:
It's true, we could probably do this without the explicit takeoff call. Can you make a pull request to remove it?

wenii, 2024-05-06:
"I tried both with the APM SITL simulator and a real drone. If there's no TAKE_OFF command in create_mission_raw, it won't take off. Also, keeping the TAKE_OFF command in the mission plan without explicitly calling the takeoff function won't make it take off either."

julianoes, 2024-05-08:
Hm, so it doesn't reset current to 0? Can you explicitly to `set_current` to reset to 0?

wenii, 2024-05-08:
Yes, later I found that setting set_current to 0 after uploading a new mission allows the new mission to take effect before the completion of the first mission. However, if the first mission has already been completed and a new mission is uploaded afterward, the new mission still may not execute. Then, I set the command of the last mission_item in the first mission to MAV_CMD_NAV_LOITER_UNLIM. This allows the second mission to execute seamlessly.

julianoes, 2024-05-08:
I'm a bit confused why it doesn't reset the current on re-upload, that's odd.

julianoes, 2025-06-02:
Sounds like this is resolved.