mavlink/MAVSDK #2310: Adding Home Position Mavlink Message to Telemetry

YasinBedirhanSimsek, 2024-05-21:
Currently, the [`subcribe_home` function](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_telemetry.html#classmavsdk_1_1_telemetry_1ae57e4c8741ba7ae05961cadfabc82875) in Telemetry API  returns a HomeHandle that uses the position struct which is in terms of global GPS position.

However, there is a mavlink message that gives more information about the [home position ](https://mavlink.io/en/messages/common.html#HOME_POSITION)

In future releases, it would be much better to implement this mavlink message into mavsdk

JonasVautherin, 2024-05-21:
Is it implemented both by PX4 and Ardupilot in a consistent way? Genuinely curious, I don't know about this HOME_POSITION message.

YasinBedirhanSimsek, 2024-05-21:
I can see it via QGroundControl (Gazebo Garden SITL)
[image]

JonasVautherin, 2024-05-21:
Sent regularly at 0.2Hz by both PX4 and Ardupilot?

YasinBedirhanSimsek, 2024-05-22:
I do not have information on Ardupilot because I do not have a test setup. On PX4 however, the answer seems to be a yes. There is also a UORB message type defined for it in the docs https://docs.px4.io/main/en/msg_docs/HomePosition.html