mavlink/MAVSDK #2316: Steps to v3

julianoes, 2024-05-27:
These API breaking issues I have planned for v3:

- [ ] https://github.com/mavlink/MAVSDK/issues/2308
- [ ] https://github.com/mavlink/MAVSDK/issues/2314
- [ ] https://github.com/mavlink/MAVSDK/issues/2315

Pjort, 2024-11-28:
Is there any chance that v3 can ship without all the "Nice to have"? 

Since to me it looks like the only thing holding back the v3 release is "multi camera support" which is in the "Nice to have" category. And this has been the case for 4-5 months now I believe.

julianoes, 2024-11-28:
I have done the bulk of the refactoring work, and it's ultimately what I need v3 for, so I guess it's more than nice to have for me.

It's now just pending system_tests failing on macOS. Since I don't have a Mac, I rented a Mac in the cloud to try to reproduce the issue. And guess what, the tests work just fine there, so it's something odd with GitHub actions mac runners, very frustrating...

Pjort, 2024-11-30:
> I have done the bulk of the refactoring work, and it's ultimately what I need v3 for, so I guess it's more than nice to have for me.
>
> It's now just pending system_tests failing on macOS. Since I [...]

I see. I have an MacBook available to me, and I can help test stuff if needed. But if it is a GitHub actions problem I am not sure how I can help.

julianoes, 2024-12-02:
Thanks @Pjort. It presumably only happens on macOS CI because it's somehow running slow or in bursts. In any case @JonasVautherin stepped in to help and was able to somewhat reproduce and fix it!

julianoes, 2024-12-18:
Getting closer.

Pjort, 2024-12-18:
> Getting closer.

Can't wait 😁

julianoes, 2025-01-17:
https://github.com/mavlink/MAVSDK/releases/tag/v3.0.0

closed 2025-01-17 by julianoes