mavlink/MAVSDK #2319: MAV_RESULT_IN_PROGRESS: Sigfault when multiple IN_PROGRESS messages are sent before arming

RaubCamaioni, 2024-05-28:
Drone Device: Skydio X2

When sending arming commands to the Skydio X2, it performs a long warm-up routine.
While the warm-up routine is running, MAV_RESULT_IN_PROGRESS messages are sent.
When the drone is finished warming up it returns a MAV_RESULT_ACCEPTED message.

This state flow is represented in MAVSDK/src/mavsdk/core/mavlink_command_sender.cpp.

Whenever the second MAV_RESULT_IN_PROGRESS is sent, I get a Sigfault from mavsdk.
It occurs when freeing the memory of a promise. 
 
The "call_callback(temp_callback, temp_result.first, temp_result.second);" function inside  
"MavlinkCommandSender::receive_command_ack(mavlink_message_t message)" is the start of the error stack.  

Does the dev team have any drones that return MAV_RESULT_IN_PROGRESS messages before success when arming?
Is this an expected working feature. 
The workflow does not work for the Skydio X2.

Inside MAVSDK/src/mavsdk/core/mavlink_command_sender.cpp:
All other case statement flows remove the job from the queue: "_work_queue.erase(it)"
Likely, any job that reaches the call_callback function twice will cause a Sigfault. 
The only jobs able to go through the case statement twice are jobs that are returned a PROGRESS message.

Current mitigation: Disabling call_callback for any Progress message. 
Mavlink seems to work perfectly fine with call_callback disabled for Progress message, must not be used anywhere crucial. 

```
        if (temp_callback != nullptr) {
            if (temp_result.first != Result::InProgress)
            {
                call_callback(temp_callback, temp_result.first, temp_result.second);
            }
        }
```

julianoes, 2024-05-29:
I think you're right. We should ignore `InProgress` here:

https://github.com/mavlink/MAVSDK/blob/23517a00c9395cae8ddf5a31cbd81596cc08309a/src/mavsdk/plugins/action/action_impl.cpp#L246

Do you want to create a pull request? We should probably fix it for all commands in `ActionImpl`.

For some background, this has not come up because PX4 and ArduPilot don't send `ACK_IN_PROGRESS` which has really only been adopted later by cameras etc.

RaubCamaioni, 2024-05-29:
"We should probably fix it for all commands in ActionImpl"
My understanding of the mavsdk repo is too limited to make an intelligent fix. 

I can test any provided fix and confirm PROGRESS messages do not cause a Sigfault anymore.

julianoes, 2024-05-30:
This turned out to be quite easy to fix: https://github.com/mavlink/MAVSDK/pull/2320

I haven't tested it, so it would be great if you could give it a quick check.

closed 2024-05-30 by julianoes

RaubCamaioni, 2024-06-28:
> This turned out to be quite easy to fix: #2320
> 
> I haven't tested it, so it would be great if you could give it a quick check.

Checked the newest release of mavsdk on pipy.
Everything worked as expected.  Good work.

PR #2320 (2024-05-30): action: ignore InProgress updates
We can't call the callback more than once, otherwise we segfault with "promise already satisfied".

Fixes #2319