mavlink/MAVSDK #2328: Unable to install latest version 2.12.0-debian11-arm64 on jetson jetpack 5.1 docker

PavelGuzenfeld, 2024-06-03:
Reproduction steps:
install latest package on my docker with the following versions:

```
$ uname -a
Linux dcs 5.10.104-tegra #1 SMP PREEMPT Thu Jul 20 07:48:15 CEST 2023 aarch64 aarch64 aarch64 GNU/Linux

$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
```

Error trace
```
$ dpkg -i ./libmavsdk-dev_2.12.0_debian11_arm64.deb
dpkg: error processing archive ./libmavsdk-dev_2.12.0_debian11_arm64.deb (--install):
 package architecture (arm64-lts) does not match system (arm64)
Errors were encountered while processing:
 ./libmavsdk-dev_2.12.0_debian11_arm64.deb
```

julianoes, 2024-06-03:
Funny, thanks for the note. I will have to fix this.

julianoes, 2024-06-03:
Fix coming: https://github.com/mavlink/MAVSDK/pull/2329

closed 2024-06-03 by julianoes