mavlink/MAVSDK #2334: libmavsdk-dev_2.12.1_debian12_arm64 - fails to link (cmake)

PavelGuzenfeld, 2024-06-04:
Following #2328 fix, cmake fails to link.

Last tested mavsdk working version - `libmavsdk-dev_2.0.1_debian11_arm64.deb`

Error trace:
```
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_create', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_join', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_detach', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_once', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to '__libc_single_threaded', version 'GLIBC_2.32'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_cond_clockwait', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'fstat64', version 'GLIBC_2.33'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'pthread_mutex_trylock', version 'GLIBC_2.34'
/usr/lib/libmavsdk.so.2.12.1: error: undefined reference to 'stat64', version 'GLIBC_2.33'
```

Investigation trace:
Docker trace:
```
$ uname -a
Linux dcs 5.10.104-tegra #1 SMP PREEMPT Thu Jul 20 07:48:15 CEST 2023 aarch64 aarch64 aarch64 GNU/Linux

$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"

$ ldd --version
ldd (Ubuntu GLIBC 2.31-0ubuntu9.14) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

The issue seems to be `glibc` version=2.31 and `debian11`.

**Kindly add back support for `debian11`.**

julianoes, 2024-06-04:
I don't think CI worked:

[image]

There is no artifact that I can see. I think the renaming didn't work and they are overwritten. I'll have to give it another go.

julianoes, 2024-06-04:
Let's see if this is fixed with https://github.com/mavlink/MAVSDK/releases/tag/v2.12.2.

PR #2335 (2024-06-04): CI: fix arm64 lib getting overwritten
Closes #2334.
closed 2024-06-04 by julianoes