mavlink/MAVSDK #2336: mavsdk_server can't discover system

Hmilzy, 2024-06-06:
Hello !
I have a raspiberry pi ,  it linked with Pixhawk 6X throught serial.           rapi: serial0 -> ttyAMA0
I have set some parameters using QGC like this : 
1. MAV_1_CONFIG = TELE 2
2. MAV_1_MODE = Onboard
3. MAV_1_RATE = 5000
4. MAV_1_FORWARD = Disabled
5. SER_TEL2_BAUD = 921600

Firmware Version : 1.14.0
mavsdk_server Verson: 1.4.17

I deploy mavsdk_server in raspi, but it seems not discover mav_system. Do you have any advices ? Thank you !

./mavsdk_server_linux-arm64-musl -p 50040 serial:///dev/ttyAMA0:921600

[image]

julianoes, 2024-06-06:
That all looks correct. 

Have you checked whether any bytes are arriving? E.g. using screen? If not, it would be some electrical issue. Make sure Tx and Rx are crossed over (so tx goes to rx and vice versa), and make sure GND is also connected.

You are using an old version of MAVSDK (and PX4). Is that on purpose?

julianoes, 2025-05-06:
No answer, closing.

closed 2025-05-06 by julianoes