mavlink/MAVSDK #2337: MavSDK server is connected to drone but not listening to port

Shkryob, 2024-06-10:
I am trying to get telemetry from my drone connected to Windows PC.

I have started MavSDK server and it is connected to drone. I can see warning messages from drone:

```
PS C:\Projects> .\mavsdk_server_win32.exe -p 50059 serial:///COM6
[11:45:51|Info ] MAVSDK version: v2.12.2 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:26)
[11:45:51|Info ] Waiting to discover system on serial:///COM6... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
[11:45:51|Debug] New system ID: 1 Comp ID: 250 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:720)
[11:45:51|Debug] Component Unsupported component (250) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:366)
[11:45:51|Warn ] Vehicle type changed (new type: 2, old type: 0) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:217)
[11:45:51|Debug] Discovered 1 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:509)
[11:47:41|Debug] MAVLink: notice: THROTTLE IS NOT LOW
```

But I can't connect python script to server:

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    # Init the drone
    drone = System(mavsdk_server_address='127.0.0.1', port=50051)
    print('Drone connection start')
    await drone.connect()

    # Start the tasks
    print('Tasks start')
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_gps_info(drone))
    asyncio.ensure_future(print_in_air(drone))
    asyncio.ensure_future(print_position(drone))
```

Prints out:

```
PS C:\Projects\sauron-eye> & C:/Python311/python.exe c:/Projects/sauron-eye/main.py
Drone connection start
```

NetCat doesn't see port open either:

```
PS C:\Windows\system32> nmap localhost -p 50051
Starting Nmap 7.95 ( https://nmap.org ) at 2024-06-10 11:45 FLE Daylight Time
Nmap scan report for localhost (127.0.0.1)
Host is up (0.00s latency).
Other addresses for localhost (not scanned): ::1
rDNS record for 127.0.0.1: kubernetes.docker.internal

PORT      STATE  SERVICE
50051/tcp closed unknown

Nmap done: 1 IP address (1 host up) scanned in 0.15 seconds
```

julianoes, 2024-06-10:
I don't know. Does maybe the mavsdk_server not have "network" access. Sometimes you have to like enable it in the firewall. I have seen this "access to internet" popup come up on Windows when you lauch it, in certain configurations.

Shkryob, 2024-06-10:
@julianoes is it ok that it says `Component **Unsupported** component (250) added.` ?

julianoes, 2024-06-10:
Oh, that might explain it. Why is the compid 250? It's supposed to be 1 for an autopilot.

closed 2025-05-20 by julianoes