mavlink/MAVSDK #2339: MAVSDK Download Log is always error with missing chunk ??

aminballoon, 2024-06-12:
if have use MAVSDK 2.12.2 Download Log feature bit while its downloading its always show 
```
[03:11:56|Error] Timeout! (log_files_impl.cpp:380)
[03:11:56|Error] Requesting missing chunk:      513/757 (log_files_impl.cpp:381)
[03:12:03|Error] Timeout! (log_files_impl.cpp:380)
[03:12:03|Error] Requesting missing chunk:      599/757 (log_files_impl.cpp:381)
[03:12:04|Error] Timeout! (log_files_impl.cpp:380)
[03:12:04|Error] Requesting missing chunk:      600/757 (log_files_impl.cpp:381)
[03:12:11|Error] Timeout! (log_files_impl.cpp:380)
[03:12:11|Error] Requesting missing chunk:      674/757 (log_files_impl.cpp:381)
[03:12:17|Error] Timeout! (log_files_impl.cpp:380)
[03:12:17|Error] Requesting missing chunk:      729/757 (log_files_impl.cpp:381)
```
What happen with this ??

julianoes, 2024-06-14:
Can you add steps how to reproduce this? What link is this over?

aminballoon, 2024-06-20:
What information does you want to know more about this issue ??

julianoes, 2024-06-22:
What code do you run? E.g. an example? How is the Pixhawk connected? USB? Is it on Linux, macOS, Windows? Just more information. Which PX4 version?

aminballoon, 2024-06-23:
How is the Pixhawk connected > "UART/TTL" on "Linux", i have tried PX4 1.12.3, 1.14.3 its always get report  missing chunk , Its can downloaded log but I want to know what happen of this error .

dakejahl, 2024-08-21:
@aminballoon chunks can get missed if your link isn't 100% lossless. Perhaps this would be a warning rather than an error

julianoes, 2025-02-12:
@git-lisx ok, what MAVSDK version is this and which mavsdk_server version (if different from what is shipped with it)?

julianoes, 2025-06-02:
I don't know how to reproduce this, so I'm closing it.

closed 2025-06-02 by julianoes