mavlink/MAVSDK #2340: mission download on the server side

heiheiheihhhhh, 2024-06-14:
Hi,If I am not directly connected to PX4 or AUTOPILOT, does the mission download on the server side have an interface?

julianoes, 2024-06-14:
I don't understand what you mea. Can you give a bit more context about your setup and what you're trying to do?

julianoes, 2024-06-15:
Aha, so on the server (so autopilot side) you need to use the [MissionRawServer](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw_server/include/plugins/mission_raw_server/mission_raw_server.h) API.

We call upload from ground station to autopilot, by the way.

Also, if you use server plugins, make sure to use MAVSDK v2. Server plugins were not well supported with MAVSDK v1.4.

heiheiheihhhhh, 2024-06-16:
> Aha, so on the server (so autopilot side) you need to use the [...]

Thank you for your reply. The situation is that I can currently implement the flight paths upload from the ground station to autopilot. I want to implement the flight paths download from the autopilot to the ground station and have found the mission_download API in the mission plugin, but I am not sure how to provide the flight paths to the ground station.

julianoes, 2024-06-19:
Plugins used on the autopilot side are called `_server`. So that's what you want to use on that side, generally.

heiheiheihhhhh, 2024-06-19:
> Plugins used on the autopilot side are called . So that's what you want to use on that side, generally.`_server`

But there is no download_mission API in mission_raw_server plugin.

julianoes, 2024-06-19:
No, the way it works is that the client (e.g. a ground station) does the upload or download. The server just responds and does the transfer.

heiheiheihhhhh, 2024-06-20:
> No, the way it works is that the client (e.g. a ground station) does the upload or download. The server just responds and does the transfer.

I have tried using the mission_download API of the mission plugin on the client side, but it returned an empty MissionPlan.

julianoes, 2024-06-22:
So there was no mission saved? Sorry, it's difficult to understand what is going on. You might have to share a code snippet or reproducible example.

heiheiheihhhhh, 2024-07-01:
> So there was no mission saved? Sorry, it's difficult to understand what is going on. You might have to share a code snippet or reproducible example.

I apologize for the delayed response. I have reviewed the MAVLink official website regarding the download sequence diagram and found that in MAVSDK, there is only the send of MISSION_REQUEST_LIST, with no subscribe. In other words, MAVSDK indeed lacks the communication for download on the server side. Therefore, I have added a class called DownloadServerWorkItem to the mavlink_mission_transfer framework to solve this problem.
[image]
[image]

julianoes, 2024-07-03:
I think you're misunderstanding this. The server doesn't initiate or do a mission download. It only "serves". If you want to download a mission, you would use the mission_raw plugin.

heiheiheihhhhh, 2024-07-04:
> I think you're misunderstanding this. The server doesn't initiate or do a mission download. It only "serves". If you want to download a mission, you would use the mission_raw plugin.

I have tried to directly call mission_upload and mission_download in the mission plugin, but I did not obtain the MissionPlan. Below are parts of my source code and the results of execution.
[image]
[image]
[image]

julianoes, 2024-07-04:
Ok, that would be a bug. Can you share code for me to reproduce and also mention PX4 or ArduPilot version that you're using this against?