mavlink/MAVSDK #2346: Disable automatically re-sending setpoints in offboard

hugh-cover-ms, 2024-07-08:
Is there a way to disable the MAVSDK offboard class from automatically re-sending setpoints?

It seems that this is done to bypass the fact that Px4 needs off-board setpoints to be minimally sent at 2Hz, however there is a good reason Px4 does that. 
The current behaviour of MAVSDK can leave the drone mindlessly following the last setpoint if there is an unexpected fault on the companion computer. This is particularly a problem with velocity setpoints.

I guess I could modify the value of SEND_INTERVAL_S, but that seems like a less than optimal solution.

julianoes, 2024-07-08:
> if there is an unexpected fault on the companion computer.

What would that be? I feel like such a fault should not happen, and if it happens it should tear down MAVSDK cleanly.

julianoes, 2025-05-06:
No more response, closing.

closed 2025-05-06 by julianoes