mavlink/MAVSDK #2347: MAVSDK SERVO_OUTPUT_RAW(36) DATA Receive

boytimo, 2024-07-09:
Hello,

I have a question. I am using the Pixhawk Orange Cube Plus and MAVSDK to communicate between the Pixhawk and a Raspberry Pi. While I have successfully received GPS data, roll, pitch, yaw angles, and various other data, I have not been able to receive servo data. The following code is causing an error:

cpp
const Telemetry::Result set_rate_actuator_output_result = telemetry.set_rate_actuator_output_status(1.0);
if (set_rate_actuator_output_result != Telemetry::Result::Success)
{
    std::cerr << "Setting rate failed: " << set_rate_actuator_output_result << '\n';
    return 1;
}
The error code is MAVLink: info: No ap_message for mavlink id (375) (system_impl.cpp:243), and when I connect the Pixhawk to Mission Planner and inspect the MAVLink messages, the list shows:

Vehicle 1
 L Comp 1 MAV_COMP_ID_AUTOPILOT1
  L SERVO_OUTPUT_RAW (2.0 Hz, #36) 48Bps
    L port 0 system.Byte
    L servo1_raw 0 system.UInt16
    ...
    L time_usec x system.UInt32
According to the official documentation, the SERVO_OUTPUT_RAW ID has been replaced by ACTUATOR_OUTPUT_STATUS. Is it possible to modify the MAVSDK code to receive SERVO_OUTPUT_RAW? If not, are there any other solutions?

Thank you very much.

[image]

julianoes, 2024-07-11:
PX4 or ArduPilot? And which version?

The message you're trying to set the rate is: [ACTUATOR_OUTPUT_STATUS](https://mavlink.io/en/messages/common.html#ACTUATOR_OUTPUT_STATUS).

Presumably ArduPilot doesn't implement that one?

Could you just use MavlinkPassthrough instead for this one?

boytimo, 2024-07-16:
I solved it with MavlinkPassthrough. Thank you for your reply.

closed 2024-07-17 by julianoes