mavlink/MAVSDK #2352: QUESTION: Forwarding RTK RTCM data to Flight Controller using the RTK Plugin with .sendRtcmData(), using MAVSDK-Java

Ba0Nguyen, 2024-07-16:
I am receiving ByteArrays of RTCM Data from a service provider and I want to forward this data to the Flight Controller.
I used the similar code to the below example: 
'
val rtcmEncodedString: String = Base64.encodeToString(buffer, Base64.NO_WRAP)
                    val rtkData = Rtk.RtcmData(rtcmEncodedString)
                    val completable = rtk.sendRtcmData(rtkData)
                    Log.d("rtk", "Rtk Data: ${rtkData.data}")
                    completable.subscribe ({
                        Log.d("rtk", "Forward RTK data completed")
                    }, { e ->
                        Log.d("rtk", "Error: ${e.message}")
                    })
'

For testing, I am still seeing GpsInfo showing 3D Locked instead of RTK, any help would be appreciated. Thanks!

Ba0Nguyen, 2024-07-16:
I also received Success responses in forwarding the RTCM Data.

julianoes, 2024-07-17:
The current release v2.12.2 does not use base64 encoding yet.
See: https://github.com/mavlink/MAVSDK/blob/v2.12.2/src/mavsdk/plugins/rtk/rtk_impl.cpp

The next one will though:
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/rtk/rtk_impl.cpp#L33

Ba0Nguyen, 2024-07-17:
I tried with the current version by converting ByteArray to String using:
`val rtcmDataString = String(bufferArray)
val RtcmData = RtcmData(rtcmDataString)`
I was able to forward the RtcmData Successfully but GpsInfo still show 'Lock 3D'

I ended up using the rtk_impl.cpp for Base64 encode and it works. I am now seeing RTK Locked from GpsInfo. 

Thanks!

closed 2024-07-17 by Ba0Nguyen