mavlink/MAVSDK #2353: TCP connection from MAVSDK breaks app after 16 minutes of idle time due to reconnection

fibonacci-matrix, 2024-07-19:
I'm developing an app that uses MAVSDK to connect to a device via the built-in MAVSDK TCP connection. MAVSDK has an excellent TCP reconnection mechanism that fits my requirements well, namely the possibility of temporary disconnection. In this case timer is started, which checks every second whether the connection has been restored. But after 16 minutes, this mechanism breaks my app with the message **"socket errorToo many open files"**. 

I did some basic research and the problem is that the MAVSDK in a loop reopens the sockets, but does not close them with the `close` function in case of unsuccessful execution of the `connect` function. 

```
// from tcp_connection.cpp

_socket_fd = socket(AF_INET, SOCK_STREAM, 0);
```

I have tested basic fix on Linux, but I don't know if it will work on other operating systems

JonasVautherin, 2024-07-19:
Would you like to open a PR with your proposed fix? :blush:

julianoes, 2024-07-20:
Good catch. Yes a PR would be great.

fibonacci-matrix, 2024-07-22:
I can do a PR in a few days, but I don't mind if someone else does it. Besides, https://github.com/mavlink/MAVSDK/pull/2342 will conflict with my changes

julianoes, 2024-07-22:
#2342 is merged.

PR #2357 (2024-07-25): core: Correctly close sockets
Closes: #2353
closed 2024-07-25 by julianoes