mavlink/MAVSDK #2355: Example error ：[ImportError: sys.meta_path is None, Python is likely shutting down]

XiaJunfly, 2024-07-22:
When I using MAVSDK to run an example, such as mission.py, the following error occurs. What is the problem? Who can help me! Thanks!
`
Waiting for drone to connect...
-- Connected to drone!
Traceback (most recent call last):
  File "C:\Users\33165\Desktop\fly vision\examples\mission_import.py", line 34, in <module>
    asyncio.run(run())
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\33165\Desktop\fly vision\examples\mission_import.py", line 20, in run
    out = await drone.mission_raw.import_qgroundcontrol_mission(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\mission_raw.py", line 1093, in import_qgroundcontrol_mission
    raise MissionRawError(result, "import_qgroundcontrol_mission()", qgc_plan_path)
mavsdk.mission_raw.MissionRawError: FAILED_TO_OPEN_QGC_PLAN: 'Failed To Open Qgc Plan'; origin: import_qgroundcontrol_mission(); params: ('example-mission.plan',)
Exception ignored in: <function System.__del__ at 0x00000143E8074360>
Traceback (most recent call last):
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 88, in __del__
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 124, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down
`

XiaJunfly, 2024-07-22:
[image]

JonasVautherin, 2024-07-22:
Apparently something is going wrong with the QGC plan you are trying to import:

> mavsdk.mission_raw.MissionRawError: FAILED_TO_OPEN_QGC_PLAN: 'Failed To Open Qgc Plan'; origin: import_qgroundcontrol_mission(); params: ('example-mission.plan',)

julianoes, 2024-07-22:
Could it be that it doesn't find the path? Try with an absolute path. It's tricky because I think the path is relative to where the mavsdk_server binary is running, and not the Python script.

XiaJunfly, 2024-07-23:
> Apparently something is going wrong with the QGC plan you are trying to import:

Thanks, it works. However, other example shows the same error, but it's not import any .plan files, for example:`mission.py`, shows:
"""
Waiting for drone to connect...
-- Connected to drone!
-- Uploading mission
Waiting for drone to have a global position estimate...
Mission progress: 0/3
-- Global position estimate OK
-- Arming
-- Starting mission
Traceback (most recent call last):
  File "C:\Users\33165\Desktop\fly vision\examples\mission.py", line 124, in <module>
    asyncio.run(run())
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\33165\Desktop\fly vision\examples\mission.py", line 88, in run
    await drone.mission.start_mission()
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\mission.py", line 1199, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: DENIED: 'Denied'; origin: start_mission(); params: ()
Exception ignored in: <function System.__del__ at 0x0000018B1C513B00>
Traceback (most recent call last):
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 88, in __del__
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 124, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down
"""
Can you help me to slove this problem？Thankyou again.

XiaJunfly, 2024-07-23:
> Could it be that it doesn't find the path? Try with an absolute path. It's tricky because I think the path is relative to where the mavsdk_server binary is running, and not the Python script.

Thanks, it works. However, other example shows the same error, but it's not import any .plan files, for example:`mission.py`, shows:
"""
Waiting for drone to connect...
-- Connected to drone!
-- Uploading mission
Waiting for drone to have a global position estimate...
Mission progress: 0/3
-- Global position estimate OK
-- Arming
-- Starting mission
Traceback (most recent call last):
  File "C:\Users\33165\Desktop\fly vision\examples\mission.py", line 124, in <module>
    asyncio.run(run())
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\python\anaconda3\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\33165\Desktop\fly vision\examples\mission.py", line 88, in run
    await drone.mission.start_mission()
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\mission.py", line 1199, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: DENIED: 'Denied'; origin: start_mission(); params: ()
Exception ignored in: <function System.__del__ at 0x0000018B1C513B00>
Traceback (most recent call last):
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 88, in __del__
  File "D:\python\anaconda3\Lib\site-packages\mavsdk\system.py", line 124, in _stop_mavsdk_server
ImportError: sys.meta_path is None, Python is likely shutting down
"""
Can you help me to slove this problem？Thankyou again.

XiaJunfly, 2024-07-23:
> Apparently something is going wrong with the QGC plan you are trying to import:
> 
> > mavsdk.mission_raw.MissionRawError: FAILED_TO_OPEN_QGC_PLAN: 'Failed To Open Qgc Plan'; origin: import_qgroundcontrol_mission(); params: ('example-mission.plan',)

Thanks, I'm try with an absolute path,it works.

XiaJunfly, 2024-07-23:
> Could it be that it doesn't find the path? Try with an absolute path. It's tricky because I think the path is relative to where the mavsdk_server binary is running, and not the Python script.

Thanks, lt works.

closed 2024-07-23 by XiaJunfly