mavlink/MAVSDK #2362: odometry data is incorrect

0725-wen, 2024-07-29:
"When I retrieve the odometry data, I find that the data is inaccurate and the errors vary greatly, making it difficult to use the offboard mode to control the drone. Is this an issue with the PX4 flight controller or MAVSDK? Below, I'll post the odometry data I retrieved."
[image]
[image]
[image]

below is my code to get odometry data:
```
 while(1){
        Telemetry::Odometry odom = telemetry.odometry();
        std::cout << "odom: frameid:" << odom.frame_id << "x:" << odom.position_body.x << ", y:" << odom.position_body.y << ", z:" << odom.position_body.z << std::endl;
        std::this_thread::sleep_for(std::chrono::milliseconds(100));
    }
```

julianoes, 2024-07-29:
Can you give some context what you're trying to do? And what versions you're using?

0725-wen, 2024-07-29:
I am using offboard mode to control a drone, which requires odometry data. However, during testing, I found that the data is inaccurate, making it impossible to precisely control the drone. We are using PX4 firmware version 14.3, MAVSDK version 2.6.0, and the flight controller hardware is Pixhawk 6C. The odometry data printed above is from a test case I wrote. Ideally, when the drone is stationary and powered on, the body_end's x, y, z should be (0, 0, 0). However, this is not the case, and the x, y, z values are fluctuating.

0725-wen, 2024-07-29:
> Can you give some context what you're trying to do? And what versions you're using?

I am using offboard mode to control a drone, which requires odometry data. However, during testing, I found that the data is inaccurate, making it impossible to precisely control the drone. We are using PX4 firmware version 14.3, MAVSDK version 2.6.0, and the flight controller hardware is Pixhawk 6C. The odometry data printed above is from a test case I wrote. Ideally, when the drone is stationary and powered on, the body_end's x, y, z should be (0, 0, 0). However, this is not the case, and the x, y, z values are fluctuating.

julianoes, 2024-07-29:
So, what's generating the odometry data? Isn't odometry for a vision system that provides that?

julianoes, 2025-05-06:
No more answes, closing.

closed 2025-05-06 by julianoes