mavlink/MAVSDK #2365: Camera: support more than one camera with one plugin instantiation

julianoes, 2024-08-04:
In order to support more than one camera from language wrappers, we need to change the camera API, so it supports more than one camera with just the one instance.

This would be a similar change to the API as the [gimbal API change](https://github.com/mavlink/MAVSDK/pull/2363).

Do you have objections @JonasVautherin?

JonasVautherin, 2024-08-05:
In the past I wanted to make it such that one could instantiate multiple `Camera()` objects (one for each), instead of calling each function with a camera id. Unfortunately gRPC does not seem to make it very convenient, so I guess the id makes sense :+1:.

julianoes, 2024-12-18:
Done in https://github.com/mavlink/MAVSDK/pull/2386.

closed 2024-12-18 by julianoes