mavlink/MAVSDK #2367: connected with a real drone problem

wangpuxia, 2024-08-09:
hi

I set up a mavsdk environment on the server and tested it using jmavsim. Everything work fine
But when I connected the real drone to Mavsdk, Some functions can be used normally, such as

System drone = new System ( "123.201.28.156" , 50041 ) ;
drone.getTelemetry().getFlightMode().subscribe(flightMode -> logger.info("Flight mode: " + flightMode));

It can return the flightmode data correctly, but most functions such as

drone.getAction ( ).arm ( ) or
drone.getTelemetry ( ).getBattery ( )

do not provide any feedback. The server will output content similar to the following:

[09:49:05|Info ] MAVSDK version: v2.11.0 (mavsdk_impl.cpp:26)
[09:49:05|Info ] Waiting to discover system on udp://:14552... (connection_initiator.h:20)
[09:50:20|Info ] New system on: 39.144.112.2:48334 (with system ID: 1) (udp_connection.cpp:199)
[09:50:20|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:720)
[09:50:20|Debug] Component Autopilot (1) added. (system_impl.cpp:366)
[09:50:20|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[09:50:20|Debug] Discovered 1 component(s) (system_impl.cpp:509)
[09:50:20|Info ] System discovered (connection_initiator.h:62)
[09:50:20|Info ] Server started (grpc_server.cpp:169)
[09:50:20|Info ] Server set to listen on 0.0.0.0:50042 (grpc_server.cpp:170)
[09:50:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:50:22|Debug] MAVLink: info: u-blox 1 HW: 00080000 SW: EXT CORE 3.01 (107900) (system_impl.cpp:243)
[09:50:51|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:51|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:51:17|Warn ] sending again after 0.502024 s, retries to do: 3 (512). (mavlink_command_sender.cpp:320)
[09:51:17|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:51:18|Warn ] sending again, retries to do: 5 (mavlink_parameter_client.cpp:1057)
[09:51:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:51:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:51:51|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:51:51|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:52:13|Warn ] sending again after 0.501734 s, retries to do: 3 (512). (mavlink_command_sender.cpp:320)
[09:52:13|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:52:21|Warn ] sending again after 0.51101 s, retries to do: 3 (512). (mavlink_command_sender.cpp:320)
[09:52:21|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:326)
[09:52:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:52:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:52:50|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:52:50|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:53:20|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:53:20|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:53:50|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:53:50|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)
[09:54:20|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:54:20|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)

May I ask where the problem occurred? and how to fix it.

JonasVautherin, 2024-08-09:
>  [09:50:21|Debug] MAVLink: critical: PreArm: Throttle below Failsafe (system_impl.cpp:243)
[09:50:21|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:243)

This suggests that the `arm` command reaches the autopilot, and the autopilot refuses to arm (because of the throttle). Are you sure that MAVSDK doesn't return an error there? How do you actually call it (`drone.getAction().arm()` won't call it, you have to subscribe at some point).

For the telemetry, are you sure that your drone is sending those updates? E.g. does QGC receive them?

julianoes, 2024-08-09:
Duplicate of https://github.com/mavlink/MAVSDK-Java/issues/174

closed 2024-08-09 by julianoes