mavlink/MAVSDK #2374: Build From Source Failure (Raspbian OS)

codymarquardt, 2024-08-15:
I have a raspberry pi 4 running Raspbian (Linux 11 Bullseye). I want to build mavsdk from source so that I can have full control over debugging and making changes to the source. 

I followed the tutorial from (https://mavsdk.mavlink.io/main/en/cpp/guide/build.html) and have run cmake to get the errors shown below. The output log is also shown below. I have tried building the mavlink third_party library separately and that was not successful either. Am I not going to be able to build this on Raspbian?

`cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
-- Version: v2.12.2-71-g3b5f3dc2
-- Version major: 2
-- Version minor: 12
-- Version patch: 2
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/MAVSDK/third_party/mavlink
Error: could not load cache
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  /home/pi/MAVSDK/build/default/third_party/mavlink failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:11 (build_target)

-- Configuring incomplete, errors occurred!
See also "/home/pi/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
`

cat build/default/CMakeFiles/CMakeOutput.log 
The system is: Linux - 5.10.33-v7l+ - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/MAVSDK/build/default/CMakeFiles/3.18.4/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/pi/MAVSDK/build/default/CMakeFiles/3.18.4/CompilerIdCXX/a.out"

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake cmTC_12011/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_12011.dir/build.make CMakeFiles/cmTC_12011.dir/build
gmake[1]: Entering directory '/home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_12011.dir/CMakeCCompilerABI.c.o
/usr/bin/cc   -v -o CMakeFiles/cmTC_12011.dir/CMakeCCompilerABI.c.o -c /usr/share/cmake-3.18/Modules/CMakeCCompilerABI.c
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
Target: arm-linux-gnueabihf
[... 338 lines cut ...]
    arg [/usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/crti.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/10/crtbegin.o] ==> ignore
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/10] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/10]
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf]
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/10/../../..] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/10/../../..]
    arg [-L/lib/arm-linux-gnueabihf] ==> dir [/lib/arm-linux-gnueabihf]
    arg [-L/usr/lib/arm-linux-gnueabihf] ==> dir [/usr/lib/arm-linux-gnueabihf]
    arg [CMakeFiles/cmTC_f9e2a.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/arm-linux-gnueabihf/10/crtend.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/crtn.o] ==> ignore
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/10] ==> [/usr/lib/gcc/arm-linux-gnueabihf/10]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/10/../../..] ==> [/usr/lib]
  collapse library dir [/lib/arm-linux-gnueabihf] ==> [/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit dirs: [/usr/lib/gcc/arm-linux-gnueabihf/10;/usr/lib/arm-linux-gnueabihf;/usr/lib;/lib/arm-linux-gnueabihf]
  implicit fwks: []

JonasVautherin, 2024-08-15:
> Error: could not load cache

This sounds like it's not a clean build. Is this the output of a clean build?

codymarquardt, 2024-08-15:
I cleaned off the build from scratch and that fixed the load cache error. Now, it is giving me this

`[ 37%] Performing configure step for 'libevents'
CMake Error at common/CMakeLists.txt:3 (add_library):
  add_library INTERFACE library requires no source arguments.

CMake Error at common/CMakeLists.txt:7 (set_target_properties):
  set_target_properties Can not find target to add properties to:
  libevents_common

CMake Error at common/CMakeLists.txt:8 (install):
  install TARGETS given target "libevents_common" which does not exist.

CMake Error at parse/CMakeLists.txt:23 (add_library):
  add_library INTERFACE library requires no source arguments.

CMake Error at parse/CMakeLists.txt:27 (set_target_properties):
  set_target_properties Can not find target to add properties to:
  nlohmann_json

CMake Error at parse/CMakeLists.txt:29 (install):
  install TARGETS given target "nlohmann_json" which does not exist.

CMake Error at protocol/CMakeLists.txt:3 (add_library):
  add_library INTERFACE library requires no source arguments.

CMake Error at protocol/CMakeLists.txt:7 (set_target_properties):
  set_target_properties Can not find target to add properties to:
  libevents_receive

CMake Error at protocol/CMakeLists.txt:8 (install):
  install TARGETS given target "libevents_receive" which does not exist.

-- Enabling format target
-- Configuring incomplete, errors occurred!
See also "/home/pi/MAVSDK/build/default/third_party/libevents/libevents-prefix/src/libevents-build/CMakeFiles/CMakeOutput.log".
gmake[2]: *** [CMakeFiles/libevents.dir/build.make:125: libevents-prefix/src/libevents-stamp/libevents-configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/libevents.dir/all] Error 2
gmake: *** [Makefile:103: all] Error 2
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  /home/pi/MAVSDK/build/default/third_party/libevents failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:36 (build_target)

-- Configuring incomplete, errors occurred!
See also "/home/pi/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".`

JonasVautherin, 2024-08-15:
Which version of CMake are you running?

codymarquardt, 2024-08-15:
version 3.18.4

JonasVautherin, 2024-08-15:
It seems like you need at least [CMake 3.19](https://cmake.org/cmake/help/latest/command/add_library.html#interface-with-sources) because of [this](https://github.com/mavlink/libevents/blob/main/libs/cpp/common/CMakeLists.txt#L3-L5) :shrug:

codymarquardt, 2024-08-15:
I suppose I could try installing cmake 3.19 from source. Is that the easiest in your opinion?

JonasVautherin, 2024-08-15:
Not for source. Just download a binary from https://cmake.org/download/ :blush:.

julianoes, 2024-08-16:
Also, this doesn't look right. Have you installed some sort of cross compiler GCC manually?

```
#include "..." search starts here:
#include <...> search starts here:
/usr/lib/gcc/arm-linux-gnueabihf/10/include
/usr/local/include
/usr/include/arm-linux-gnueabihf
```

codymarquardt, 2024-08-16:
I am using BlueOS, so I cannot confirm if it is or not

By updating cmake to 3.19, that fixed my issue there. There now seems to be an issue with GTest::gmock. I installed libgmock-dev already

[ 75%] Completed 'libevents'
[100%] Built target libevents
-- Found curl via config file: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL
-- All plugins enabled
-- BUILD_MAVSDK_SERVER not set: not building grpc mavsdk_server
-- Configuring done
CMake Error at src/integration_tests/CMakeLists.txt:5 (add_executable):
  Target "integration_tests_runner" links to target "GTest::gmock" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at src/unit_tests/CMakeLists.txt:3 (add_executable):
  Target "unit_tests_runner" links to target "GTest::gmock" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at src/system_tests/CMakeLists.txt:1 (add_executable):
  Target "system_tests_runner" links to target "GTest::gmock" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

codymarquardt, 2024-08-16:
https://discuss.px4.io/t/mavsdk-main-branch-fails-to-build-on-rpi4-blueos/40003/9

I found a post discussing this exact same issue on the PX4 forum with no fix

JonasVautherin, 2024-08-18:
Can you try building MAVSDK without the tests? I made a PR here to allow that: https://github.com/mavlink/MAVSDK/pull/2377.

You should be able to build with `-DBUILD_TESTING=OFF` :+1:.

codymarquardt, 2024-08-18:
Hi Jonas, 

That is actually what I ended up doing!

Thanks for the help. I think we can close this

closed 2024-08-18 by julianoes