mavlink/MAVSDK #2375: Connecting my MAVSDK server with c++ to my front-end

VinayAquaairx, 2024-08-17:
Hi @JonasVautherin  @julianoes  This is vinayak i have front end ready and i installed jmavsim and MAVSDK_server, and grpc and connected these with my front-end the data is coming but now i want to try with my drone and i dont know how to connect i coonected my drone with lan to my pc and i tried my rasberry py ./build/default/src/mavsdk_server/src/mavsdk_server udp://:192.168.2.2:14550 -p 50000 this is my ipaddress and port address and i am getting this error 

[06:16:04|Info ] MAVSDK version: v2.12.2-55-g249b064e (mavsdk_impl.cpp:28)
[06:16:04|Info ] Waiting to discover system on udp://:192.168.2.2:14550... (connection_initiator.h:20)
[06:16:04|Warn ] Connection using udp:// is deprecated, please use udpin:// or udpout:// (cli_arg.cpp:28)
[06:16:04|Error] bind error: Permission denied (udp_connection.cpp:84)
[06:16:04|Error] Connection failed: Bind error (connection_initiator.h:47)

can you help me connection my drone which sends data to my front-end

Thank You
vinayak

julianoes, 2024-08-18:
First of all, you are not using the latest release but you are using main which actually changes the connection API. However, that's not documented yet. So be aware of that. This warning actually tells you:

```
 [06:16:04|Warn ] Connection using udp:// is deprecated, please use udpin:// or udpout:// (cli_arg.cpp:28)
```

What you have to know is that with UDP connections, there are two ways:

- udpout (previously `udp://IP:PORT`) -> Send MAVLink messages to this IP and port.
- udpin (previously `udp://:PORT`) -> Listen for MAVLink on all network interfaces on this port. Someone else needs to send the traffic there.

I don't quite understand your setup and how it all connects. To help further I'd have to know how it all connects.

julianoes, 2024-08-18:
And make sure to read through this:
https://mavsdk.mavlink.io/main/en/cpp/guide/connections.html

VinayAquaairx, 2024-08-19:
Hey first i installed JMAVSIM simulation to get sample data from drone and installed MAVSDK server from https://mavsdk.mavlink.io/main/en/cpp/guide/build.html this by following those steps and next i installed grpc which gets data from mavsdk server and it sends the data to my front-end this is what i did. Now i installed https://github.com/mavlink/MAVSDK/releases the latest version and followed these steps https://mavsdk.mavlink.io/main/en/cpp/guide/build.html to build it. Now so if there is another way for getting data through mavsdk server to front-end. please let me know and can you please tell me the steps what to do or any

[image] to be clear i followed these steps upto now.

julianoes, 2024-08-19:
Right, what I mean is how the things connect.

julianoes, 2025-05-06:
No more answers in a while, I'm closing this.