mavlink/MAVSDK #2376: Issue with action.terminate() in terminate.cpp example with SITL

KarthiAru, 2024-08-18:
Hi,

I am currently working with the terminate.cpp example in SITL using gazebo and the gazebo_typhoon_h480 models. I am encountering an issue where the `action.terminate()` function does not seem to work as expected.

**Expected Behavior:**
-  `action.terminate()` should disarm the drone and open the parachute, causing the drone's altitude to decrease and the drone to disarm.

**Actual Behavior:**
-  The `action.terminate()` function does not result in the drone disarming or deploying the parachute. The altitude remains unchanged, and I receive an error message related to a poll timeout.
-  The `action.kill()` function, on the other hand, works as intended, causing the drone’s altitude to drop to 0 and the drone to disarm.

Log: [https://review.px4.io/plot_app?log=a7744711-82ea-42ae-9be7-6bc1f615d787](https://review.px4.io/plot_app?log=a7744711-82ea-42ae-9be7-6bc1f615d787) _(Note: I triggered RTL at the end for the purpose of saving the log file.)_

I have also attempted to enable the parameter COM_PARACHUTE, but the issue persists. Are there additional parameters or configurations required to ensure `action.terminate()` functions correctly?

KarthiAru, 2024-08-18:
I reviewed the [action plugin implementation](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L331) and found that it uses the MAVLink command [MAV_CMD_DO_FLIGHTTERMINATION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_FLIGHTTERMINATION). However, I am unsure how to proceed with troubleshooting this issue further.

KarthiAru, 2024-08-18:
I tried terminating in the middle of a mission. It transitioned to an unknown flight mode in QGC.

[image]

Log: [https://review.px4.io/plot_app?log=978520ae-5dcb-47c2-b182-3361e679d96f](https://review.px4.io/plot_app?log=978520ae-5dcb-47c2-b182-3361e679d96f)

julianoes, 2024-08-18:
> an error message related to a poll timeout

I believe this is a SITL problem. The issue is that all actuator outputs are stopped, and so the lockstep simulation is hanging, because there are no more outputs being published. You could argue that it is working as intended but termination is presumably not correctly implemented in SITL.

You could try without lockstep, see: https://docs.px4.io/main/en/simulation/#disable-lockstep-simulation

KarthiAru, 2024-08-19:
@julianoes Thanks. It works by setting the `<enable_lockstep>false</enable_lockstep>` in the model SDF file. Closing this since it is not a MADSDK issue.

closed 2024-08-19 by KarthiAru