mavlink/MAVSDK #2378: Connecting to Ardusub system

codymarquardt, 2024-08-19:
I am running Ardusub on a bluerov2. I am trying to call mavsdk.add_any_connection() and cannot find a connection with a valid system. 

From docs, I see it is udpin://0.0.0.0:14550, but that’s not working. 

Is there any confirmation that this works on a BlueROV? I can’t find any resources on others doing this. Unsure if this is a sdk issue or lack of documentation issue

julianoes, 2024-08-19:
Where does BlueRov2 send or listen to MAVLink?

codymarquardt, 2024-08-19:
[image]

This is what I can see on the mavlink endpoints page of BlueOS

julianoes, 2024-08-19:
The first one looks like it would be listening on all network interfaces (0.0.0.0), so if MAVSDK is running on the same host, it would connect to that one using `udp://127.0.0.1:14550`.

julianoes, 2025-05-06:
No more question/comments, closing.