mavlink/MAVSDK #2389: error building for ARM

ben1tdsx, 2024-09-02:
I'm trying to build for Nvidia Jetson Nano and I'm getting the following error:

[100%] Linking CXX executable integration_tests_runner
/usr/bin/ld: integration_tests_runner: hidden symbol `__aarch64_swp1_acq_rel' in /usr/lib/gcc/aarch64-linux-gnu/10/libgcc.a(swp_1_4.o) is referenced by DSO
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
src/integration_tests/CMakeFiles/integration_tests_runner.dir/build.make:837: recipe for target 'src/integration_tests/integration_tests_runner' failed
make[2]: *** [src/integration_tests/integration_tests_runner] Error 1
CMakeFiles/Makefile2:917: recipe for target 'src/integration_tests/CMakeFiles/integration_tests_runner.dir/all' failed
make[1]: *** [src/integration_tests/CMakeFiles/integration_tests_runner.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

JonasVautherin, 2024-09-02:
Can you try disabling the tests? `-DBUILD_TESTING=OFF`

ben1tdsx, 2024-09-03:
thanks, I was able to build and install mavsdk successfully.

closed 2024-09-03 by ben1tdsx