mavlink/MAVSDK #2390: "undefined reference" linker error with ARM build

ben1tdsx, 2024-09-03:
I'm able to successfully build MAVSDK with the following cmake command

cmake -DCMAKE_CXX_COMPILER=g++-10 -DCMAKE_EXE_LINKER_FLAGS="-latomic" -DBUILD_TESTING=OFF ..

But when I try to build a project that links against libmavsdk, I'm getting the following linker errors

/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_ldset4_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_ldadd8_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_cas4_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_ldadd2_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_swp4_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_ldadd4_acq_rel'
/usr/local/lib/libmavsdk.so.2.12.2: undefined reference to `__aarch64_swp1_acq_rel'

julianoes, 2024-09-03:
Can you try this?
https://stackoverflow.com/a/75291816/8548472

ben1tdsx, 2024-09-04:
I changed the compiler to

set(CMAKE_CXX_COMPILER g++-10)

and most of the linker errors go away. Only one is being reported now.

/usr/bin/ld: ArduRoverControl: hidden symbol `__aarch64_ldset4_acq_rel' in /usr/lib/gcc/aarch64-linux-gnu/10/libgcc.a(ldset_4_4.o) is referenced by DSO

Did you mean to try to rebuild libmavsdk with "-mno-outline-atomics" or add the flag to the CMakeLists.txt file for the project I'm trying to build that links against libmavsdk? 

I tried to add the "-mno-outline-atomics" flag to my project that links against libmavsdk and I still get the "hidden symbol" linker error.

julianoes, 2024-09-04:
The linking error seems to happen in your own code ArduRoverControl? No?

ben1tdsx, 2024-09-05:
Yes, the linking error is happening in my code. It is not happening when attempting to build libmavsdk. I'm able to successfully build/install libmavsdk.

julianoes, 2024-09-06:
Right, so it's not a libmavsdk issue?

ben1tdsx, 2024-09-06:
I think so, I also tried the QuickStart guide:

https://mavsdk.mavlink.io/main/en/cpp/quickstart.html

And when I try to "sudo dpkg..." the .deb file I get the following error:

dpkg: error processing archive libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb (--install):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb

Anyway, I noticed that there's a docker folder in the repo, but I can't find any docker build instructions. Is docker the preferred method for building libmavsdk for arm64?

ben1tdsx, 2024-09-06:
Looks like the run-docker script doesn't support arm64.

sudo ./run-docker-clang-format.sh
Unable to find image 'mavsdk/mavsdk-clang-format:latest' locally
latest: Pulling from mavsdk/mavsdk-clang-format
9c704ecd0c69: Pull complete 
06e936d8a7cd: Pull complete 
5f67a7f1391d: Pull complete 
Digest: sha256:8df530b2cd8340895063792cdd04cd0e53bc815b9f3b3189994a4436779a03ec
Status: Downloaded newer image for mavsdk/mavsdk-clang-format:latest
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
exec /bin/bash: exec format error

julianoes, 2024-09-07:
Why are you trying to install amd64 on arm?

ben1tdsx, 2024-09-07:
thanks for catching that. I was just blindly using the example shell commands to install the pre-built library. 

Jetson Nano can support 32bit ARMv7 libraries, but the Ubuntu image Nvidia provides does not support 32bit libraries by default. I had to do this:

sudo dpkg --add-architecture armhf
sudo apt update
sudo apt install libc6:armhf

before I can install with this:

wget https://github.com/mavlink/MAVSDK/releases/download/v2.12.5/libmavsdk-dev_2.12.5_debian12_armv7.deb
sudo dpkg -i libmavsdk-dev_2.12.5_debian12_armv7.deb

I also had to add some linker instructions to my CMakeLists.txt file to force the linker to use gcc_s instead of the static version. 

target_link_libraries(ArduRoverControl
    MAVSDK::mavsdk
    atomic
    gcc_s
)

set_target_properties(ArduRoverControl PROPERTIES
    LINK_FLAGS "-shared-libgcc -latomic -fuse-ld=gold"
)

I'm now able to successfully link libmavsdk!! 

Thanks for all the help.

ben1tdsx, 2024-09-07:
BTW, I was assuming there isn't an ARMv8 prebuilt version of libmavsdk since I only saw arm6 and arm7 at https://github.com/mavlink/MAVSDK/releases

closed 2024-09-07 by ben1tdsx

julianoes, 2024-09-07:
Right, it's called arm64.