mavlink/MAVSDK #2392: mavsdk_server doesn't check if the server it is connected to is MavSDK.

AlexisTM, 2024-09-04:
## Problem:

If another gRPC server is already on port `50051`, MavSDK connect function succeeds, while the mavsdk_server starts and silently fails in the background.

For us it was `QIKService3`, some Qualcomm service for board discovery, which was listening on `tcp6://0.0.0.0:50051`.

Ideally, there should be a verification the server we connected to is valid.  

## To reproduce

Instal `QIKService3` or a gRPC test server on 50051.
I tried netcat listening on port 50051 without success, so I suppose  it probably needs to be a gRPC server.

```python
import asyncio
import logging
import sys
from mavsdk import System

logging.basicConfig(stream=sys.stdout, level=logging.DEBUG)

drone = System()
await drone.connect(system_address="udp://10.151.60.12:14550")
# This succeeds even if no drone is connected because a server listen on 50051
version = await drone.info.get_version()
# This causes an error because get_version is not implemented.
# mavsdk_server dies silently in the background 
```

## Log

By default, the logging does not report those errors.

```
>>> drone = System()
>>> await drone.connect(system_address="udp://10.151.60.12:14550")
DEBUG:mavsdk.async_plugin_manager:Waiting for mavsdk_server to be ready...
DEBUG:mavsdk.system:[01:07:14|Info ] MAVSDK version: v2.12.5 (mavsdk_impl.cpp:26)
DEBUG:mavsdk.async_plugin_manager:Connected to mavsdk_server!
DEBUG:mavsdk.system:E0904 13:07:14.466524237   87615 chttp2_server.cc:1060]                UNKNOWN:No address added out of total 1 resolved for '0.0.0.0:50051' {created_time:"2024-09-04T13:07:14.466032843+02:00", children:[FAILED_PRECONDITION:Failed to add any wildcard listeners: Error in bind for address '[::]:50051': Address in useError in bind for address '0.0.0.0:50051': Address in use]}
DEBUG:mavsdk.system:[01:07:14|Info ] Waiting to discover system on udp://10.151.60.12:14550... (connection_initiator.h:22)
DEBUG:mavsdk.system:[01:07:14|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:720)
DEBUG:mavsdk.system:[01:07:14|Debug] Component Autopilot (1) added. (system_impl.cpp:366)
DEBUG:mavsdk.system:[01:07:14|Debug] MAVLink: info: Connection to ground station lost	(system_impl.cpp:243)
DEBUG:mavsdk.system:[01:07:14|Info ] System discovered (connection_initiator.h:38)
DEBUG:mavsdk.system:[01:07:14|Error] Failed to bind server to port 50051 (grpc_server.cpp:172)
DEBUG:mavsdk.system:Failed to start, exiting...
>>> version = await drone.info.get_version()
Traceback (most recent call last):
  File "/usr/lib/python3.12/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "<console>", line 1, in <module>
  File "/home/alexis/mavsdk_venv/lib/python3.12/site-packages/mavsdk/info.py", line 856, in get_version
    response = await self._stub.GetVersion(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alexis/mavsdk_venv/lib/python3.12/site-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
           ^^^^^^^^^
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2024-09-04T13:07:16.212002745+02:00", grpc_status:12, grpc_message:""}"
```

julianoes, 2024-09-04:
What do you suggest we do?

AlexisTM, 2024-09-05:
I am not sure. 
It took me a long time to figure out what was happening. A system identification would do the trick, do a gRPC call which ensure we are connected to a _mavsdk_ server?

JonasVautherin, 2024-09-06:
> A system identification would do the trick, do a gRPC call which ensure we are connected to a mavsdk server?

Would it, though? If `mavsdk_server` starts successfully, then you will eventually receive MAVLink messages (that's your system identification). If `mavsdk_server` fails, you won't receive any MAVLink, but your gRPC "identification" call will fail just the same.

As you say, `mavsdk_server` knows that the port is already bound, and fails:

```
DEBUG:mavsdk.system:[01:07:14|Error] Failed to bind server to port 50051 (grpc_server.cpp:172)
DEBUG:mavsdk.system:Failed to start, exiting...
```

So to me, your problem here seems to be that somehow you did not have access to the `mavsdk_server` logs. Any kind of automation we would put on top of this would result in you getting the error message where you want it (because if the port you want to use for `mavsdk_server` is already bound, there is no way it will ever work).

Maybe my question would be: why did it take a long time for you to access the `mavsdk_server` logs, and do you think there is something that could make it easier? My feeling is that it is a bit of a leaky abstraction: users have to understand that there is a server running under the hood, so that if it doesn't work they immediately go check its logs :thinking:.

What do you think? :innocent:

azerupi, 2024-09-06:
>Maybe my question would be: why did it take a long time for you to access the mavsdk_server logs, and do you think there is something that could make it easier?

It is not obvious because `mavsdk_server` is being spawned by the python `connect()` call but even though `mavsdk_sever` doesn't bind to the port and exits the python code goes on as if it was connected. Then after that any call to `mavsdk_server` results in an `UNIMPLEMENTED` error which doesn't really give a good hint of what is going on.
It is only when you enable debug logs that you can find the `mavsdk_server` error within all the other output.

If `connect()` would check that `mavsdk_server` was successfully running and throw an exception otherwise instead of continuing as if everything is fine, that would solve the issue here I believe. 

Another option that would help is if we can promote `mavsdk_server` errors to loglevel error instead of debug. Then they would show up without having to change the loglevel I assume?

Yet another option would be to check the port before launching `mavsdk_server` and warn or throw an exception if it is not free.

Do any of those make sense to you?

azerupi, 2024-09-06:
>though I assume that users may end up turning the MAVSDK logs off (the issue here exists because at least you guys don't keep them enabled all the time, so it seems like a fair assumption 😁).

As far as I can tell the `mavsdk_server` logs in MAVSDK-Python are only printed as debug logs and debug logs are not shown by default. So the `mavsdk_server` logs end up being swallowed and we didn't see the `mavsdk_server` error. This is not something we explicitly disabled.

While I don't necessarily have a great solution, the point I was trying to make was that anything that would prevent `connect()` from silently continuing as if `mavsdk_server` was working would have saved us time in figuring out the issue. And while we have figured out the problem I assume someone else will eventually hit the same issue. If we can make it easier for the next guy to figure it out it is a worthwhile investment :slightly_smiling_face: 

>Hmm it feels like a hack to me 😅. But nothing prevents you from doing it from your code, before you start MAVSDK!

I don't disagree, this solution is a bit of a hack. But to the second part, I don't think it is fair to expect the user to perform this check because the `mavsdk_server` and port are implementation details. We definitely were not aware MAVSDK-Python was using a grpc server before hitting and looking deeper into this issue.

>Maybe there is something to do with the return code of mavsdk_server? Like MAVSDK-Python could detect that mavsdk_server stopped (with or without an error code) and... I don't know... throw an exception in Python? But then again, the exception would say "something wrong happened with mavsdk_server, please check the logs" 🙈.

Yes, logging if the process exits before we expect it to would be a step in the right direction and most likely sufficient. Do you have any objections to this solution? Personally I feel like it would have made the error more explicit and given us a hint were to look.

JonasVautherin, 2024-09-09:
> Yes, logging if the process exits before we expect it to would be a step in the right direction and most likely sufficient. Do you have any objections to this solution?

No, I think it would make sense for MAVSDK-Python to monitor `mavsdk_server` when it runs it :+1:.

AlexisTM, 2024-09-18:
That would be enough

AlexisTM, 2024-09-18:
Note that looking on the web, 50051 seems to be the default for many gRPC servers thus could happen to others:
- https://xrpl.org/docs/infrastructure/configuration/configure-grpc
- https://weaviate.io/developers/weaviate/api/grpc

Quick question, shouldn't we first check if a mavsdk_server exists before starting our own? Isn't mavsdk used with multiple clients to a single mavsdk_sever instance? (at least that is how I understood the project)

Today, the instance we start would crash if another mavsdk_server instance is already started and we would connect to the one that existed before. If we monitor our own instance, I expect we won't be able to run multiple clients.

AlexisTM, 2024-09-20:
I will close this. It will be a good reference for the next person searching for this issue :)

closed 2024-09-20 by AlexisTM