mavlink/MAVSDK #2402: MAVSDK Ubuntu Release deb contains old file

alexcekay, 2024-09-17:
When installing _libmavsdk-dev_2.12.5_ubuntu22.04_amd64.deb_ I can't build the ftp_client example due to the following compile errors:

`home/alex/MAVSDK/examples/ftp_client/ftp_client.cpp: In function ‘mavsdk::Ftp::Result remove_directory(mavsdk::Ftp&, const string&, bool)’:
/home/alex/MAVSDK/examples/ftp_client/ftp_client.cpp:73:62: error: ‘ListDirectoryData’ is not a member of ‘mavsdk::Ftp’`

A quick investigation shows that by installing the deb the following header file is included:
_/usr/include/mavsdk/plugins/ftp/ftp.h_

The relevant lines of code are as follows in this file:
```
    /**
     * @brief Destructor (internal use only).
     */
    ~Ftp() override;

    /**
     * @brief Progress data type for file transfer.
     */
    struct ProgressData {
        uint32_t bytes_transferred{}; /**< @brief The number of bytes already transferred. */
        uint32_t total_bytes{}; /**< @brief The total bytes to transfer. */
    };
```

When building MAVSDK from source the examples work as the relevant lines of code are as follows in this file:
```
    /**
     * @brief Destructor (internal use only).
     */
    ~Ftp() override;

    /**
     * @brief
     */
    struct ListDirectoryData {
        std::vector<std::string> dirs{}; /**< @brief The found directories. */
        std::vector<std::string> files{}; /**< @brief The found files. */
    };
```

julianoes, 2024-09-17:
Can you use the v2.12 branch instead? The release is not matching main, main is going to be v3.

alexcekay, 2024-09-17:
Ahh I see that one is matching. Yes, I can use that one. Thanks :+1:

closed 2024-09-19 by alexcekay

McCoyChisom, 2024-09-27:
i specifically need libmavsdk-dev_2.10.1_ubuntu22.04_amd64.deb

julianoes, 2024-09-29:
@McCoyChisom why?
It's here https://github.com/mavlink/MAVSDK/releases/tag/v2.10.1

TsomBillzz, 2024-10-01:
Reason is that I had used v2.10.1 in my uav project earlier this year, around May. Now, I am working on an iteration of the project. Unfortunately, I crashed the project without tracking with .git. Good news, is that I still have some of my files, so I just want to rebuild the earlier project and smoothly transition to the latest iteration, without encountering some kind of dependency issues.

McCoyChisom, 2024-10-01:
> @McCoyChisom why? It's here https://github.com/mavlink/MAVSDK/releases/tag/v2.10.1

Reason is that I had used v2.10.1 in my uav project earlier this year, around May. Now, I am working on an iteration of the project. Unfortunately, I crashed the project without tracking with .git. Good news, is that I still have some of my files, so I just want to rebuild the earlier project and smoothly transition to the latest iteration, without encountering some kind of dependency issues.

McCoyChisom, 2024-10-01:
@julianoes 
Reason is that I had used v2.10.1 in my uav project earlier this year, around May. Now, I am working on an iteration of the project. Unfortunately, I crashed the project without tracking with .git. Good news, is that I still have some of my files, so I just want to rebuild the earlier project and smoothly transition to the latest iteration, without encountering some kind of dependency issues.