mavlink/MAVSDK #2403: missing function download_geofence

mattes-bru, 2024-09-20:
I am currently looking into bringing the geofence feature to an application I am developing using Mavsdk. For the mission feature I am currently downloading the mission from the autopilot on connect and on changes. With geofences this is not possible, because there is no download function and also no callback to get notified on changes.

Is there a reason for not having these functions or is there just a pr needed? If latter is the case, what would be the best way to implement them? 

This also applies to the rally point feature.

julianoes, 2024-09-21:
No reason, just no one asked for it yet :). Feel free to make a pull request.

julianoes, 2025-05-06:
Thanks for the PR, closing this.

closed 2025-05-06 by julianoes