mavlink/MAVSDK #2408: Error in `fly_mission` example: `Falling back to gimbal protocol v1 (mission_impl.cpp:128)`

jonazti07, 2024-09-30:
I'm trying to run the `fly_mission` example after building in the following way:
```
cd /path/to/MAVSDK/examples/fly_mission
cmake -Bbuild -H.
cmake --build build -j4
```

Then, I try to launch:
```
build/fly_mission udp://:5763
```

Note that I'm running the SITL of ArduPilot as explained in [here](https://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#start-sitl-simulator) and adding the port by typing `output add 127.0.0.1:5763` in the terminal.

The error I obtain while trying to run the example is as follows:
```
[04:09:51|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:128)
Mission upload failed: Unsupported, exiting.
```

julianoes, 2024-09-30:
Interesting, thanks for the issue. I will have to investigate.

julianoes, 2025-05-06:
I think this is no longer valid as we now have removed the gimbal protocol v1 to avoid these annoying edge cases.