mavlink/MAVSDK #2415: Simplify build process

dakejahl, 2024-10-04:
**Issue**
It's not super easy to build the library, and I'm not sure how to do a clean build without removing the dependencies added via FetchContent. It would be nice to be able to remove only the source build files so that we don't have to always redownload the dependencies (right now I just delete build/ and install/)

**Solution**
For my C++ projects I always add a Makefile to wrap the CMake commands. I like doing this because for every project I know I can run `make`, `make clean` and `make install`. The underlying actions of each recipe could vary, but it makes it easy for me to switch between projects without having to remember specifics of how they're built/deployed.

JonasVautherin, 2024-10-04:
I have [opinions](https://www.acarg.ch/posts/cmake-deps/) [on that topic](https://www.acarg.ch/posts/cmake-superbuild/) :grin:.

> I'm not sure how to do a clean build without removing the dependencies added via FetchContent.

We don't use FetchContent, we use ExternalProject_Add. I find the difference important because we use `find_package` where FetchContent encourages the use of `add_subdirectory()` which I find awful for dependencies.

> For my C++ projects I always add a Makefile to wrap the CMake commands.

I would strongly discourage that. In my experience it tends to end up in a weird Automake/CMake mix that nobody understands (PX4 used to be like that).

What you should be able to do (and if that doesn't work, then it needs to be fixed) is:

1. Build the third parties separately from [third_party/](https://github.com/mavlink/MAVSDK/tree/main/third_party). Something like:

```cmake
cmake -DCMAKE_INSTALL_PREFIX=/path/to/where/you/want/third/parties -DCMAKE_PREFIX_PATH=/path/to/where/you/want/third/parties -Bbuild -S.
cmake --build build --target install
```

This should build and install the dependencies in `/path/to/where/you/want/third/parties`. This only needs to be done once. You may even skip that and install the dependencies from your system package manager if that works.

This is basically what is described [here](https://www.acarg.ch/posts/cmake-deps/), but our superbuild makes it a bit more convoluted.

2. Build MAVSDK with `-DSUPERBUILD=OFF` and `-DCMAKE_PREFIX_PATH=/path/to/where/you/want/third/parties`. This will completely skip the superbuild and fetch the dependencies you built in the previous step.

[I wish we didn't have a superbuild](https://www.acarg.ch/posts/cmake-superbuild), but the fact is that most people don't know how the basics of CMake and therefore we need a superbuild. I'm already happy that we don't have the same kind of Makefile wrapping CMake that PX4 used to have :innocent:.

But MAVSDK does work without the superbuild (`-DSUPERBUILD=OFF`), and this is tested in the CI (to some extent) when we build system packages.

Does that help?

dakejahl, 2024-10-04:
I think you illustrated my point :grin: I shouldn't need to know all these details. It would be nice to have a default obvious way to build/install/clean.

> I would strongly discourage that. In my experience it tends to end up in a weird Automake/CMake mix that nobody understands

I use `make` basically as a wrapper for bash. You could use bash too **install.sh** **build.sh** **clean.sh**. The idea is a one line command that is stupid easy to remember. It would make it easier to work on the project.

dakejahl, 2024-10-04:
Anyway, my opinion... one could argue that you shouldn't work on a project without knowing exactly how it works. I end up bouncing around between a lot of projects and it's a lot easier to remember just one build/install/clean command with sane defaults that I can use for every project.

JonasVautherin, 2024-10-04:
> I shouldn't need to know all these details.

Depends on what details you mean. As I said, I don't like superbuilds (even though I wrote this one) because they only work in specific situations and are generally not standard (even Meson tries to have a first class superbuild which is admittedly nicer when you want it, but also even more annoying when you don't want it).

IMHO, if you use a C++ project that uses CMake, you should know that:

1. CMake has a configure step (e.g. `cmake -Bbuild -S.`) and a build step (e.g. `cmake --build build`).
2. CMake will try to find dependencies that are included with `find_package()`, either on your system or in a custom path if you set `CMAKE_PREFIX_PATH`.
3. The project may try to provide helpers for you to build the dependencies, e.g. in the form of a superbuild.

By having a superbuild, we make it easier for people who don't want to ever see `-DCMAKE_PREFIX_PATH`, but at the cost of making it harder for you (because now there is an indirection with `-DSUPERBUILD=OFF`).

> You could use bash too install.sh build.sh clean.sh.

Say we have that. I assume that `build.sh` builds all the dependencies + MAVSDK, and `clean.sh` removes all of those. How does that solve your problem, which is "I'm not sure how to do a clean build without removing the dependencies"?

Should we have `build.sh`, `build_dependencies_only.sh`, `build_mavsdk_only.sh`, `install.sh`, `install_dependencies_only.sh`, `install_mavsdk_only.sh`, `clean.sh`, `clean_dependencies_only.sh`, `clean_mavsdk_only.sh`, ...?

My solution is this: choose a build system (in this case we chose CMake) and use it the way it's meant to be used (no Makefile wrappers, no magic superbuilds like the one I wrote here, ...). And then agree on the fact that developers should know the two commands they need to build a CMake project (or at least how to Google for them). :innocent:

dakejahl, 2024-10-04:
Think of this issue report as coming from a "dumb user". I don't want to learn anything, I don't want to read the docs, I just want to build/install/clean in a way that could be considered the most standard way possible (you can assume I'm not modifying dependencies, because dumb users don't do that). It's not that I can't learn or remember how to do things the "standard" way for a given toolchain, I just shouldn't need to. Especially if it's a single LOC change. I think we probably have a difference of opinion here. Honestly even remembering `cmake -Bbuild -S.` is asking too much :laughing:  

I guess what I'm getting it is I personally like to be able to Do The Thing with a single command I can type in less than 2 seconds.

closed 2024-10-04 by dakejahl

JonasVautherin, 2024-10-04:
Well we could have a `build.sh` that just builds everything, and a `clean.sh` that cleans everything. It would only solve the first half of the first sentence of this issue, but we could do it.

Actually I would be happier without a superbuild and with `build.sh` and `clean.sh` instead (though I'm nervous with scripts that run `rm -rf`, and in our case cleaning is really just `rm -rf build/`). I tried to remove the superbuild [a while ago](https://github.com/mavlink/MAVSDK/compare/main...cmake-reorg) but the conclusion was that we were losing the "easy way".

I would be happier if I could say "if you just want to build everything, run `build.sh`, and if you want to have control just use it as a CMake project".

julianoes, 2024-10-05:
> @JonasVautherin: Makefile wrapping CMake that PX4 used to have

Still has it, and I don't like it.

> @dakejahl: I shouldn't need to know all these details

I agree, except:
1. Dependencies are hard, and C++ doesn't have support for this, so it's up to the build system you choose or whatever else you do. We opted for cmake which means you need to learn/understand cmake. Yes, it's not just one command, but that's because we don't just support one OS or one setup.
2. And that is the second point. We could use something like vcpkg, conan, hunter, etc. and that might simplify things, however, as far as I'm aware that doesn't work for cross compilation builds that we support using dockcross, iOS, etc.. If we didn't have any cross-compilation it likely could be simpler.

> sane defaults that I can use for every project.

That's exactly what it would be: it would be defaults **for you** (and probably me) but not for everyone and so it can also create confusion. Having a `Makefile` would suddenly mean people will try to just use make, not realizing it's cmake, so I wouldn't want that. What I could see is a `build_linux.sh` sort of thing. However, you will end up adding arguments to this (MAVSDK bespoke) script that you will have to remember (e.g. build type). Over time you end up with this script encoding all sorts of cmake options, at which point we could just use [cmake-presets](https://cmake.org/cmake/help/latest/manual/cmake-presets.7.html), and maybe that would be worthwhile. Last time I checked it wasn't quite supported for the cmake version we were targetting but by now it could be ok. I don't think I'd use it personally, mostly because it's yet another layer on top of cmake when I can just use the cmake configure lines that I have in my zsh history.

> @JonasVautherin I do that, too. That's why I want projects to use their tools in a standard way, so that I need to learn how to use Autotools, Meson, CMake, SwiftPkg, Carthage or Gradle only once, instead of having to learn a new build.sh for each project (which may call a Makefile wrapping CMake with custom functions that call Python scripts below).

That's exactly it. You put it better than I could @JonasVautherin.

> @dakejahl I don't want to learn anything, I don't want to read the docs

Sorry, you have to read the one page on how to build it, and take around 10s to type it out. I'm sorry we don't know how to make it simpler. (And no a git submodule mess is not going to happen.)

A dumb user should install the package from deb. A contributor needs to know the 3 arguments to cmake.

julianoes, 2024-10-05:
Here is another take on why I don't want a Makefile:

If I open the project in an IDE, say VSCode or CLion, I want it to detect that it's just cmake, and it will just work. If it also has a Makefile, it will ask you, and you need to know which one is the correct option. A developer new to it will be confused and not know which one to use.

And PX4 as an example: there is duplicated cmake logic in two places because of the "easy" Makefile:

1. https://github.com/PX4/PX4-Autopilot/blob/4215e20120706221ce6f4cd0fab332a55f3bbeac/Makefile#L178-L196
2. https://github.com/PX4/PX4-Autopilot/blob/4215e20120706221ce6f4cd0fab332a55f3bbeac/.vscode/cmake-variants.yaml#L4-L8

dakejahl, 2024-10-05:
Ah yeah didn't think about non-linux. I don't use an IDE so I didn't think about the IDE auto-detect case. I guess it's not easy to implement a general solution. Anyway I only brought this up because it's a minor pain point for me, I thought if it's annoying for me it's probably annoying for others and thought it would be a good suggestion... wasn't trying to criticize or spark a debate.

I'll just keep using my out of tree Makefile :+1:

julianoes, 2024-10-05:
Fair enough. I understand where you're coming from and actually appreciate you bringing up friction points! This one is unfortunately one where it's not so easy, and we have opinions :).

JonasVautherin, 2024-10-05:
> I thought if it's annoying for me it's probably annoying for others and thought it would be a good suggestion... wasn't trying to criticize or spark a debate.

And I didn't want to sound too opinionated, sorry if that's how I came across :see_no_evil:. I guess my point is that I spent a lot of time trying different things and I haven't found a silver bullet :shrug:. Even modern language package managers come with their issues (often making it easier for developers but terrible for distro maintainers and security in general).