mavlink/MAVSDK #2420: Timeout for commands with ACK

unjambonakap, 2024-10-07:
```
D1007 02:07:07.516869 83912 system.py:53] [02:07:07|Debug] Sent command 241 (mavlink_command_sender.cpp:428)
D1007 02:07:07.524811 83912 system.py:53] [02:07:07|Debug] Got timeout! (mavlink_command_sender.cpp:298)
D1007 02:07:07.524998 83912 system.py:53] [02:07:07|Warn ] sending again after 0.00809172 s, retries to do: 3  (241). (mavlink_command_sender.cpp:322)
D1007 02:07:07.535593 83912 system.py:53] [02:07:07|Debug] Got timeout! (mavlink_command_sender.cpp:298)
D1007 02:07:07.535791 83912 system.py:53] [02:07:07|Warn ] sending again after 0.0187866 s, retries to do: 2  (241). (mavlink_command_sender.cpp:322)
D1007 02:07:07.545903 83912 system.py:53] [02:07:07|Debug] Got timeout! (mavlink_command_sender.cpp:298)
D1007 02:07:07.546084 83912 system.py:53] [02:07:07|Warn ] sending again after 0.0291914 s, retries to do: 1  (241). (mavlink_command_sender.cpp:322)
D1007 02:07:07.556502 83912 system.py:53] [02:07:07|Debug] Got timeout! (mavlink_command_sender.cpp:298)
```
Please note the time between the sends and timeouts (~instant)

It appears to be due to `_timeout_s` with a default of 0, changed recently from 0.5. Hence me wondering if this was intended or not

> Is this intended?
> 
> 0 timeout seems to be causing issues on my side 

 _Originally posted by @unjambonakap in [e889188](https://github.com/mavlink/MAVSDK/commit/e8891889d62b6d50ac413adc58fc27f9099a0dc9#r147635724)_

julianoes, 2024-10-07:
Thanks @unjambonakap, presumably that was a mistake. Oh, now I see it! Maybe it didn't load correctly before? In any case, let's fix it.

unjambonakap, 2024-10-07:
Thanks for getting back so quickly
I'll make a PR in a bit!

julianoes, 2024-10-07:
Already done, sorry :)

PR #2421 (2024-10-07): core: fix timeout
I must have broken this in a previous refactor.

Fixes #2420.
closed 2024-10-07 by julianoes